YosysHQ/yosys #1265: Multiple drivers in carry cell

olofk, 2019-08-07:
After upgrading yosys to 0.8+634 (e9a756aa) from a version built some time in March 2019 I'm getting

    ERROR: multiple drivers on net 'CO' (carry.CO and $logic_or$/usr/local/bin/../share/yosys/ice40/cells_sim.v:141$8.Y)

in nextpnr. Example design [servant_0.tar.gz](https://github.com/YosysHQ/yosys/files/3478830/servant_0.tar.gz)

To reproduce

    make -C servant_0/tinyfpga_bx-icestorm

daveshah1, 2019-08-07:
If your version of nextpnr is from before late June, you will need to update to cope with the new whitebox support in Yosys.

olofk, 2019-08-07:
Ah ok. I'll do that. Is there any particular order I need to compile the tools (yosys, nextpnr, icestorm) or is it just enough to update and rebuild nexpnr?

daveshah1, 2019-08-07:
If you haven't updated for a while, it would be a good idea to update icestorm first (as nextpnr uses icestorm to build its database), although I don't think there have been any significant changes in the last few months.

olofk, 2019-08-08:
Updating nextpnr worked fine with the added benefit of raising Fmax almost 30%. Thanks

closed 2019-08-08 by olofk