YosysHQ/yosys #1268: Unexpected if behavior

ZipCPU, 2019-08-08:
## Steps to reproduce the issue

See the attached design.

[mcve.zip](https://github.com/YosysHQ/yosys/files/3482211/mcve.zip)

Run sby -f miter.sby.

It contains two expressions for the same logic.  Depending upon a flag within, `THIS_WORKS`.  When the flag is set, the logic does what's expected.  When the flag is clear, it does not.

## Expected behavior

I expect the design to pass both ways.

## Actual behavior

When using Verific, the design passes.  Otherwise it design fails indicating that the various 'if' structures are not properly parsed.

This may be a similar or identical issue to #1243 

Dan

whitequark, 2019-08-09:
Can you recheck with master?

ZipCPU, 2019-08-14:
Just checked against master now.  The bug remains.

Dan

cliffordwolf, 2019-08-14:
TL;DR: Look at the end of this comment for a small MCVE.

I had a quick look at this. Here's a writeup of what I did: First I slightly reduced the test case, just so it fits on one screen page. Here is my `test.sv`:

```
module test (input clock, ctrl, din);
        wire foo_dout, bar_dout;

        foo foo_inst (clock, ctrl, din, foo_dout);
        bar bar_inst (clock, ctrl, din, bar_dout);

        reg check = 0;
        always @(posedge clock) check <= 1;
        always @(*) if (check) assert(foo_dout == bar_dout);
endmodule

module foo (input clock, ctrl, din, output reg dout);
        always @(posedge clock) begin
                if (1'b1) begin
                        if (1'b0) begin end else begin
                                dout <= 0;
                        end
                        if (ctrl)
                                dout <= din;
                end
        end
endmodule

module bar (input clock, ctrl, din, output reg dout);
        always @(posedge clock) begin
                if (1'b1) begin
                        if (1'b0) begin end else begin
                                dout <= 0;
                        end
                end
                if (ctrl)
                        dout <= din;
        end
endmodule
```

And here is my `test.sby`:

```
[options]
mode bmc

[engines]
smtbmc

[script]
read -noverific
read -sv test.sv
prep -top test

[files]
test.sv
```

Obviously `foo` and `bar` should be equivalent, but it fails BMC.

Looking closer, `yosys -p 'proc; opt_clean; show foo' test.sv` produces this:

[image]

But `yosys -p 'proc; opt_clean; show bar' test.sv` produces this:

[image]

The `foo` circuit looks correct, but `bar` obviously does not.

Playing a bit more with it, I arrived at the following MCVE:

```
module mcve (input clock, ctrl, din, output reg dout);
        always @(posedge clock) begin
                if (1) if (1) dout <= 0; 
                if (ctrl) dout <= din;
        end
endmodule
```

With the same result from `yosys -p 'proc; opt_clean; show' mcve.sv` as `bar` above:

[image]

whitequark, 2019-08-19:
This is interesting: `proc_clean` transforms this:
```
  process $proc$bug.v:2$1
    assign $0\dout[0:0] \dout
    attribute \src "bug.v:3"
    switch 1'1
      attribute \src "bug.v:3"
      case 1'1
        attribute \src "bug.v:4"
        switch 1'0
          attribute \src "bug.v:4"
          case 
            assign $0\dout[0:0] 1'0
        end
        attribute \src "bug.v:7"
        switch \ctrl
          attribute \src "bug.v:7"
          case 1'1
            assign $0\dout[0:0] \din
          case 
        end
      case 
    end
    sync posedge \clock
      update \dout $0\dout[0:0]
  end
```
to this:
```
  process $proc$bug.v:2$1
    assign $0\dout[0:0] \dout
    assign $0\dout[0:0] 1'0
    attribute \src "bug.v:7"
    switch \ctrl
      attribute \src "bug.v:7"
      case 1'1
        assign $0\dout[0:0] \din
    end
    sync posedge \clock
      update \dout $0\dout[0:0]
  end
```
which is clearly wrong.

whitequark, 2019-08-19:
Actually, no, `proc_clean` is actually right here. The real problem is that:
```
			if (1'b0) begin end else begin
				dout <= 0;
			end
```
somehow gets transformed into
```
        attribute \src "bug.v:4"
        switch 1'0
          attribute \src "bug.v:4"
          case 
            assign $0\dout[0:0] 1'0
        end
```
in the frontend...

whitequark, 2019-08-19:
Correction to previous comments: the problem is still in `proc_clean`, but what happens is that:
```
    assign $0\dout[0:0] \dout
    attribute \src "bug.v:15"
    switch 1'1
      attribute \src "bug.v:15"
      case 1'1
        attribute \src "bug.v:16"
        switch 1'0
          attribute \src "bug.v:16"
          case 
            assign $0\dout[0:0] 1'0
        end
      case 
    end
    attribute \src "bug.v:20"
    switch \ctrl
      attribute \src "bug.v:20"
      case 1'1
        assign $0\dout[0:0] \din
      case 
    end
```
gets transformed into:
```
    assign $0\dout[0:0] \dout
    attribute \src "bug.v:20"
    switch \ctrl
      attribute \src "bug.v:20"
      case 1'1
        assign $0\dout[0:0] \din
    end
    attribute \src "bug.v:16"
    switch 1'0
      attribute \src "bug.v:16"
      case 
        assign $0\dout[0:0] 1'0
    end
```
So, the order of the inner switches is reversed.

whitequark, 2019-08-19:
Fixed in https://github.com/YosysHQ/yosys/pull/1309. Interestingly this bug has been present in Yosys since the very first commit in 2013.

PR #1309 (2019-08-20): proc_clean: fix order of switch insertion
Fixes #1268.
commit 4a942ba7 (2019-08-20): proc_clean: fix order of switch insertion.

Fixes #1268.
closed 2019-08-20 by whitequark