YosysHQ/yosys #1269: Segmentation fault: 11 while compiling with yosys

pablogs9, 2019-08-08:
## Steps to reproduce the issue
I'm trying to implement a Verilog module that writes in a Lattice UP5K SPRAM hardware core using the Yosys SB_SPRAM256KA block. Note that there are little or no documentation/examples about usage of this black box block. The main purpose is implementing an echo or delay in an audio digital system.

I have two clocks the frame clock lrclk and the bit clock bclk, note that each period of frame clock has 64 bit clocks periods.

I tried to, with a sensitivity list in the blck, cycle a read/write process in the SPRAM. I implement a state machina that:

S1: Put the input data in the input port of the RAM, enable the write_enable signal and set the writing pointer to RAM address.
S2: (Data supposed to be written) Disables write_enable signal and set the reading pointer to RAM address.
S3: (Data supposed to be loaded on output buffer of the RAM). Set the module output from the RAM output buffer and resets the state machine.
This is the module code:

```
module echo(	
	input wire bclk,
	input wire lrclk,

	input wire [DATALEN-1:0] right_in,
	output reg [DATALEN-1:0] right_out,
);

localparam ADDRLEN = 14;
localparam DATALEN = 16;

reg [ADDRLEN-1:0] rd_ptr = 0;
reg [ADDRLEN-1:0] wr_ptr = (2**ADDRLEN)/2;

reg [2:0] sm = 0;

reg wren;
reg [ADDRLEN-1:0] memaddr;
reg [DATALEN-1:0] datain;
reg [DATALEN-1:0] dataout;

SB_SPRAM256KA M1 (
    .ADDRESS(memaddr),
    .DATAIN(datain),
    .MASKWREN(4'b1111),
    .WREN(wren),
    .CHIPSELECT(1'b1),
    .CLOCK(bclk),
    .STANDBY(1'b0),
    .SLEEP(1'b0),
    .POWEROFF(1'b0),
    .DATAOUT(dataout)
  );

always @(posedge lrclk) begin
	sm <= 1;
end

always @(posedge bclk) begin
	if (sm === 1) begin
		datain <= right_in;
		wren <= 1;
		memaddr <= wr_ptr;
		sm <= 2;
	end else if (sm === 2) begin
		wren <= 0;
		memaddr <= rd_ptr;
		sm <= 3;
	end else if (sm === 3) begin
		right_out <= dataout;
		wr_ptr <= (wr_ptr + 1);
		rd_ptr <= (rd_ptr + 1);
		sm <= 0;
	end
end

endmodule
```
## Actual behavior

I expect to have errors on systhesis time or misfunctional behaving of the implementation, but I obtain this Yosis error:

```
5.11. Executing WREDUCE pass (reducing word size of cells).
Removed top 31 bits (of 32) from port B of cell main.$add$main.v:70$2 ($add).
Removed top 1 bits (of 32) from port Y of cell main.$add$main.v:70$2 ($add).
Removed top 2 bits (of 3) from FF cell main.$techmap\E1.$procdff$117 ($dff).
make: *** [main.bin] Segmentation fault: 11
```

daveshah1, 2019-08-08:
I can't reproduce this issue on latest git master with `synth_ice40 -top echo`, although without your Yosys script I can't confirm this is exactly what you were doing.

Two observations:
 - The entire design is optimised away because of multiple drivers on `sm` (this is not legal for synthesis, and FFs clocked by more than one clock have no sensible implementation in an FPGA)
 - According to the [official documentation](http://www.latticesemi.com/~/media/LatticeSemi/Documents/TechnicalBriefs/SBTICETechnologyLibrary201608.pdf) `SB_SPRAM256KA` `POWEROFF` is active low, so needs to be tied to 1'b1 for normal operation.

pablogs9, 2019-08-08:
Sorry about the reproduction issue. Here you have the complete code:

[https://github.com/pablogs9/NodeFPGA/tree/master/Software/FPGA%20Sample%20Code/Echo%20Simple](https://github.com/pablogs9/NodeFPGA/tree/master/Software/FPGA%20Sample%20Code/Echo%20Simple)

daveshah1, 2019-08-08:
I can't reproduce the issue running `make` in that folder, either.

pablogs9, 2019-08-08:
Maybe with:

`yosys -p 'synth_ice40 -top main' main.v i2s_rx.v i2s_tx.v echo.v`

what this results to you?

daveshah1, 2019-08-08:
Still can't reproduce (or see anything obviously wrong in valgrind either).

Can you run `gdb  -ex=r --args $YOSYS/yosys -p 'synth_ice40 -top main' main.v i2s_rx.v i2s_tx.v echo.v ` and tell me what the backtrace is (replace $YOSYS with where you build Yosys, you need to use the built version not the installed version to get debug symbols)?

closed 2020-03-03 by pablogs9