YosysHQ/yosys #1279: synth_ice40 -noflatten produces multiple drivers in nextpnr

olofk, 2019-08-10:
Not sure if it's a yosys or nextpnr issue, but adding -noflatten to the synth_ice40 command gives the following error in nextpnr

    ERROR: multiple drivers on net 'o_clk' (pll.PLLOUTCORE and pll.o_clk)

Example design. [noflatten.tar.gz](https://github.com/YosysHQ/yosys/files/3488788/noflatten.tar.gz)
 Run `make -C noinithacked/tinyfpga_bx-icestorm` to reproduce. Also includes the generated json file if you want to look at that in isolation

eddiehung, 2019-08-10:
Hi Olof! Great to see you on here. I don't believe nextpnr can handle non flattened netlists at the moment.

olofk, 2019-08-12:
Hi Eddie!
No worries then. I don't actually need to run the P&R in this case, so closing this. Let me know if I should open up an enhancement issue for nextpnr

closed 2019-08-12 by olofk