YosysHQ/yosys #1284: [regression #113 build] Inverted full adder output when use -ha option.

SergeyDegtyar, 2019-08-12:
## Steps to reproduce the issue
**Original test:** 
https://github.com/YosysHQ/yosys-tests/tree/master/simple/full_adder

**Test case:**
module top
(
 input x,
 input y,
 input cin,

 output A,
 output cout
 );

assign {cout,A} =  cin + y + x;

endmodule

**script:**
read_verilog ../top.v
proc
synth -top top
extract_fa -ha -v
tee -o result.log dump
synth -top top
write_verilog synth.v

## Expected behavior

'A' shouldn't be inverted.

## Actual behavior

output 'A' has an inverted behavior.

eddiehung, 2019-08-12:
Can't seem to reproduce this locally on 78b30bbb:
```
~/yosys/master/tests/ystests/simple$ PATH=~/yosys/master:$PATH bash run.sh full_adder half_adder
+ test -d full_adder
+ '[' half_adder '!=' verify -a half_adder '!=' falsify ']'
+ test -f scripts/half_adder.ys
+ rm -rf full_adder/work_half_adder
+ mkdir full_adder/work_half_adder
+ cd full_adder/work_half_adder
+ touch .start
+ echo full_adder
+ grep '.*_error'
+ '[' half_adder = verify ']'
+ '[' half_adder = falsify ']'
+ yosys -ql yosys.log ../../scripts/half_adder.ys
+ '[' 0 '!=' 0 ']'
+ '[' -f ../../../../../techlibs/common/simcells.v ']'
+ COMMON_PREFIX=../../../../../techlibs/common
+ iverilog -o testbench ../testbench.v ../../common.v synth.v ../../../../../techlibs/common/simcells.v
+ '[' 0 '!=' 0 ']'
+ '[' half_adder = falsify ']'
+ cell_failed=0
+ test -f result.log
+ expected_string=
+ expected=1
+ '[' full_adder = aigmap ']'
+ '[' full_adder = async2sync ']'
+ '[' full_adder = clk2fflogic ']'
+ '[' full_adder = clk2fflogic_latch ']'
+ '[' full_adder = clk2fflogic_mem ']'
+ '[' full_adder = dff2dffe_unmap ']'
+ '[' full_adder = flowmap ']'
+ '[' full_adder = flowmap_latch ']'
+ '[' full_adder = flowmap_mem ']'
+ '[' full_adder = fsm ']'
+ '[' full_adder = fsm_expand ']'
+ '[' full_adder = fsm_export ']'
+ '[' full_adder = fsm_opt ']'
+ '[' full_adder = fsm_recode ']'
+ '[' full_adder = fsm_unreach ']'
+ '[' full_adder = full_adder ']'
+ expected_string='$fa'
+ test -f result.log
+ grep '$fa' result.log
  cell $fa $auto$extract_fa.cc:499:run$92
  cell $fa $auto$extract_fa.cc:499:run$85
+ '[' 1 = 1 ']'
+ cell_failed=0
+ vvp -N testbench
+ grep ERROR testbench.log
+ echo FAIL
+ touch .stamp
```

eddiehung, 2019-08-12:
Looks like it was caused by this commit: https://github.com/YosysHQ/yosys/commit/6d0be8d206e8f484473ee9fb891b0317d6741974

Specifically, uncommenting this line:
https://github.com/YosysHQ/yosys/blob/78b30bbb1102047585d1a2eac89b1c7f5ca7344e/passes/techmap/abc.cc#L1786

causes the issue to go away.

eddiehung, 2019-08-12:
^^ Because ABC then chooses to map to an OAI3 cell.

Also goes away if I comment out this out instead: 
https://github.com/YosysHQ/yosys/blob/78b30bbb1102047585d1a2eac89b1c7f5ca7344e/passes/techmap/abc.cc#L1784
to block it from using this cell.

eddiehung, 2019-08-12:
I suspect issue is in `extract_fa`:
```
read_verilog top.v
proc
synth -top top
equiv_opt -assert extract_fa -ha -v
```
fails.

eddiehung, 2019-08-14:
Okay. Started looking at this some more, and I think the cause is that `extract_fa` can cope with inverted AND gate inputs, but does not account for their effect on the XOR gate inside the adder.

This is the result of `extract_fa -ha` after it extracts the first half adder:
[image]

Note that the `$_ANDNOT_` cell is equivalent to `assign Y = A & ~B`, and sources the same `$_NOT_` output as the `$_XNOR_`.

Now the broken circuit looks like this:
[image]

which shows the new `$fa.B` cell being fed through a second `$_NOT_`, as a result of undoing the inversion of `$_ANDNOT_.B`.

An XNOR gate with either of its inputs inverted is equivalent to an XOR gate, and so one with an inverted input will cancel out any output inversion (thus making it an XOR)... right?

PR #1288 (2019-08-13): Since $_ANDNOT_ is not symmetric, do not sort leaves
Fixes #1284 

@cliffordwolf: Not sure what the value of `leaves.sort()` was to begin with?
PR #1297 (2019-08-15): extract_fa: Un-inverting AND with an inverted input also inverts input to X{,N}OR
Fixes #1284 (again, again).

See: https://github.com/YosysHQ/yosys/issues/1284#issuecomment-521455881 for an explanation of why this is necessary.
closed 2019-08-15 by eddiehung