YosysHQ/yosys #1286: Yosys emits assigns with a concatenated ecpression on the LHS

byzantic, 2019-08-12:
Yosys can emit verilog that has a concatenated signal as the recipient of an assign statement.

I.e something like:

`assign {a,b} = {c,d};`

The problem is that this causes problems for other tools; in this case OpenSTA [*], which means I can't properly verify my design.

I did not see this issue until recently, and it occurred after I introduced a `hilomap` stage, possibly because that reduces the fanin of constant buses. It's difficult to provide an example because it only seems to happen under certain circumstances.

I did try adding some `opt_clean` and `clean` stages after the `hilomap`, but that didn't seem to help.

I think I saw a previous issue/communication about this somewhere (but can't find it anymore ..), where it was stated that this was legal verilog, and therefore should be OK. Well, I can see that point of view, but it just seems it would cause less trouble if such assigns were split into multiple assigns with only one signal as their recipient.

[*] OpenSTA is really great, by the way.

pepijndevos, 2019-11-12:
I'll just add that the Gowin tools also don't like post-synthesis Verilog with this style of assignment.

whitequark, 2020-04-25:
It occurred to me that Yosys already has a pass that can avoid having such assignments be emitted: simply run `splitnets -driver` before `write_verilog`. Can you confirm this works (or does not) for you?

pepijndevos, 2020-04-26:
It seems that this would *reduce* the number of splits that are done. Post-synthesis verilog in Gowin is very limited and I'm already using plain `splitnets` to make it happy. I can try this when I look at the other Gowin changes Eddie suggested. (might be a while) What I think needs to be done is slightly different, which *does* split all the nets, but assigns individual bits to the output port. The `-port` option might actually be a workaround for Gowin, as top-level output ports are single pins anyway, but this might require adjusting pin mapping to some Yosys-generated name, which is not ideal.

byzantic, 2020-04-26:
Actually, I think this was suggested when I first raised the issue. Unfortunately, it doesn't work.

Using `splitnets -driver` does not make any difference to my output netlist - I get the same errors.
Using `splitnets` by itself makes the issue worse - I get more occurrences of the problem.

I think the point is that it's _not_ a driver issue - `splitnets` is probably doing its job quite happily.

whitequark, 2020-04-26:
Oh, you're right, and I misunderstood this issue.

pepijndevos, 2020-11-25:
Awesome!