YosysHQ/yosys #1287: Provide 'official' docker images for yosys, nextpnr and icestorm

1138-4EB, 2019-08-12:
Refs: #466, #829, #1152 and #1285.

Although there are multiple docker images available (https://hub.docker.com/search?q=yosys&type=image), AFAIK there is no 'official' one which is built in some CI service managed by the maintainers of this repo/org.

@clifford mentioned https://hub.docker.com/r/openfpga/icetools in [#466](https://github.com/YosysHQ/yosys/issues/466#issuecomment-355845977), but it seems that the WIP did not take off.

In #1285, a job is added to Travis, where an image containing yosys is built. It would be straighforward to add two more dockerfiles/images, for nextpnr and icestorm, respectively.

However, the namespace in a docker registry needs to be managed, and some credentials are needed in order to push docker images from travis to dockerhub. Note that a 'machine' or 'bot' account can be used.

For example, projects ghdl and VUnit, both have separate repos for this purpose: [ghdl/docker](https://github.com/ghdl/docker), [VUnit/docker](https://github.com/VUnit/docker). Precisely, this request is related to tgingold/ghdlsynth-beta#31.

Furthermore, it'd be possible to provide multiarch images (e.g. amd64, arm32v7 and arm64v8), so ready-to-use containers are available for any user running GNU/Linux, Windows macOs, SBCs, etc.

Note that, for now, iceprog would still be required on the host. On GNU/Linux, it is probably possible to let a container configure an FPGA through USB/FTDI; but I don't expect it to work on Windows or macOS.

emilazy, 2019-08-24:
Since these are all packaged in [nixpkgs](https://github.com/NixOS/nixpkgs), I think that [Nixery](https://nixery.appspot.com/) should already provide this as something like `nixery.dev/shell/yosys/symbiyosys/nextpnr/icestorm/trellis`.

1138-4EB, 2019-08-24:
@emilazy, thanks for the reference! I didn't know about Nixery. I tried `nixery.dev/shell/yosys/symbiyosys/nextpnr/icestorm/trellis` on Docker Desktop (Windows 10) and it works! It seems to be quite slow the first time you request to pull an image. Is it because the image is built/compiled on-demand (including all the packages)?

As you said, it is not the solution I want, because I'd like to use latest/development version. Nonetheless, I think that any section about Docker in the docs should mention it.

BTW, do you know why ghdl is not available in nixpkgs? There are some 'old' commits (https://github.com/NixOS/nixpkgs/search?q=ghdl&type=Commits), but `nixery.dev/shell/ghdl` does not work.

eine, 2021-12-21:
Closing in favour of [hub.docker.com/u/yosyshq](https://hub.docker.com/u/yosyshq) and [hdl/containers](https://github.com/hdl/containers).

closed 2021-12-21 by eine