YosysHQ/yosys #1293: always blocks

moohtuh, 2019-08-14:
How do I join chat forums here :https://freenode.irclog.whitequark.org

sorry for posting here ..please delete this

koriakin, 2019-08-14:
To join the chat, you can use any IRC client and connect to the freenode network. If you don't have an IRC client, you can use the web-based one: https://webchat.freenode.net/

As for the processes, yosys doesn't guarantee any particular ordering for them.  I suspect the order you're seeing in the output is based on the internal hash values, and shouldn't be depended upon.

closed 2019-08-14 by koriakin

moohtuh, 2019-08-14:
In Verilog, all the individual always blocks in a module execute concurrently hence the order of always blocks don't really matter.

Do you think that is the reason there is no order pattern?

koriakin, 2019-08-15:
Yes, this is the reason.