YosysHQ/yosys #1311: unable to compile: missing ice40_dsp_pm.h

xobs, 2019-08-20:
## Steps to reproduce the issue

1. Download yosys
2. Run `make config-gcc`
3. Run `make`

Output: (from https://travis-ci.org/xobs/toolchain-icestorm/jobs/574305514)

```
...
[ 46%] Building passes/opt/wreduce.o
[ 46%] Building passes/opt/opt_demorgan.o
[ 47%] Building passes/opt/rmports.o
[ 47%] Building passes/opt/opt_lut.o
[ 48%] Building passes/opt/pmux2shiftx.o
[ 48%] Building passes/opt/muxpack.o
[ 48%] Building passes/pmgen/test_pmgen_pm.h
[ 48%] Building passes/pmgen/test_pmgen.o
passes/pmgen/test_pmgen.cc:30:39: fatal error: passes/pmgen/ice40_dsp_pm.h: No such file or directory
compilation terminated.
make: *** [passes/pmgen/test_pmgen.o] Error 1
```

Note that the output from that build uses a more complex `make` command, however I have reproduced it on a different machine by simply running `make`.

whitequark, 2019-08-20:
Yep, same issue on nMigen's CI: https://travis-ci.org/m-labs/nmigen/jobs/574325647

mmicko, 2019-08-20:
It should work fine with proper dependency set. Not near development
machine right now, but will make a patch bit later with this.

OBJS += passes/pmgen/test_pmgen.o
passes/pmgen/test_pmgen.o: passes/pmgen/test_pmgen_pm.h
passes/pmgen/ice40_dsp_pm.h
$(eval $(call add_extra_objs,passes/pmgen/test_pmgen_pm.h))
$(eval $(call add_extra_objs,passes/pmgen/ice40_dsp_pm.h))

emard, 2019-08-21:
I can't find required *.h files in my latest git pull

$ find yosys -name ice40_dsp_pm.h
$

mmicko, 2019-08-21:
@emard This is generated file, fix above should make sure it is created before compiling of that line.

emard, 2019-08-21:
yes it's generated really, after applying above there's another

passes/pmgen/test_pmgen.cc:31:10: fatal error: passes/pmgen/peepopt_pm.h: Nema takve datoteke ili direktorija
 #include "passes/pmgen/peepopt_pm.h"

emard, 2019-08-21:
This makes compile to continue, lets seee....

OBJS += passes/pmgen/test_pmgen.o

passes/pmgen/test_pmgen.o: passes/pmgen/test_pmgen_pm.h passes/pmgen/ice40_dsp_pm.h passes/pmgen/peepopt_pm.h

$(eval $(call add_extra_objs,passes/pmgen/test_pmgen_pm.h))
$(eval $(call add_extra_objs,passes/pmgen/ice40_dsp_pm.h))
$(eval $(call add_extra_objs,passes/pmgen/peepopt_pm.h))

mmicko, 2019-08-21:
ah, so there is one more dependency

whitequark, 2019-08-21:
I've tried to reproduce it locally but cannot for some reason, what gives?

emard, 2019-08-21:
I compiled with aboved patch but now I get this when using yosys:

ERROR: Assert `pass_register.count(pass_name) == 0' failed in
kernel/register.cc:111.

mmicko, 2019-08-21:
@emard Sorry I can not reproduce error you are getting, if you still get it after clean compile please send me on email example how to reproduce it.

cliffordwolf, 2019-08-22:
#1315 fixes this issue. Closing now.

> ERROR: Assert `pass_register.count(pass_name) == 0' failed in kernel/register.cc:111.

That's a different problem. Please file a new issue if it persist after `make clean`.

closed 2019-08-22 by cliffordwolf