YosysHQ/yosys #1323: On ECP5, a 26-bit counter synthesizes to 271 LUTs

whitequark, 2019-08-22:
## Steps to reproduce the issue

Repro: 
[build.zip](https://github.com/YosysHQ/yosys/files/3531942/build.zip)

## Expected behavior

Synthesizes to 27 LUTs (counter plus inverter).

## Actual behavior

Synthesizes to 271 LUTs, which is an order of magnitude larger.

whitequark, 2019-08-22:
Note: I haven't verified that this bitstream runs correctly because I don't have my Versa on me.

daveshah1, 2019-08-22:
FWIW, `-abc9` produces 34 LUTs as expected

whitequark, 2019-08-22:
In fact, with `-abc9`, it synthesizes to exactly 34 LUTs. I am still keeping this issue open because the default configuration (with no options) should not give grossly oversized synthesis results; for example nMigen, by policy, does not introduce such options, leaving it to upstream to decide on the best mode.

eddiehung, 2019-08-23:
If it's any consolation, `abc9` will be default in the future, once the dust has settled. There's no roadmap to improve `abc` since it is not box-aware

Dolu1990, 2019-12-30:
I think i hit a similar issue.

Default config => 

```
Info: Logic utilisation before packing:
Info:     Total LUT4s:     11278/24288    46%
Info:         logic LUTs:   9312/24288    38%
Info:         carry LUTs:    904/24288     3%
Info:           RAM LUTs:    708/12144     5%
Info:          RAMW LUTs:    354/ 6072     5%

Info:      Total DFFs:      8904/24288    36%
...
Info:          TRELLIS_SLICE:  8334/12144    68%
```

-abc9 => 

```
 Info: Logic utilisation before packing:
Info:     Total LUT4s:      9336/24288    38%
Info:         logic LUTs:   7370/24288    30%
Info:         carry LUTs:    904/24288     3%
Info:           RAM LUTs:    708/12144     5%
Info:          RAMW LUTs:    354/ 6072     5%

Info:      Total DFFs:      8904/24288    36%

...
Info:          TRELLIS_SLICE:  7083/12144    58%
```

Performance with -abc9 seem to be a bit higher, about 8 % for my single trial case.

Dolu1990, 2019-12-30:
On diamond, it is down to 4600 slices. but the ram mapping (block/distributed) isn't the same + packing variability.
Is there a way (verilog attribute) to force a given ram to be inferred using DP16KD ? It would make the comparison better.

eddiehung, 2019-12-31:
> Is there a way (verilog attribute) to force a given ram to be inferred using DP16KD ? It would make the comparison better.

https://github.com/YosysHQ/yosys/pull/1521 added the infrastructure to do this (as long as it is still a compatible memory structure that doesn't meet the min/max requirements) but only implemented this for `synth_xilinx`, feel free to extend it for your purposes and submit a PR.

whitequark, 2023-12-18:
`abc9` is now the default: https://github.com/YosysHQ/yosys/pull/4019

closed 2023-12-18 by whitequark