YosysHQ/yosys #1324: won't compile at debian

emard, 2019-08-22:
## Steps to reproduce the issue
pull latest git
make config-gcc
make clean; make

[100%] Building abc/abc-5776ad0
ERROR: ABC contains local modifications! Set ABCREV=default in Yosys Makefile!

*Provide instructions for reproducing the issue. Make sure to include
all necessary source files. (You can simply drag&drop a .zip file into
the issue editor.)*

Also, make sure that the issue is actually reproducable in current git
master of Yosys.

See https://stackoverflow.com/help/mcve for some information on how to
create a Minimal, Complete, and Verifiable example (MCVE).

Please do not waste our time with issues that lack sufficient information
to reproduce the issue easily. We will simply close those issues.

Contact https://www.symbioticeda.com/ if you need commercial support for Yosys.

## Expected behavior

*Please describe the behavior you would have expected from the tool.*

## Actual behavior

*Please describe how the behavior you see differs from the expected behavior.*

eddiehung, 2019-08-23:
This looks like you updated from a previous version. Did you try a `git clean -fxd` to blow away everything in your repository (warning, will remove *ALL* untracked files) and try building again?

emard, 2019-08-23:
I tried this and it's the same
git pull
git clean -fxd
make config-gcc
make
...same ABC error

I can try to edit ABC = default maybe it will help now
(before this didn't help)...

mmicko, 2019-08-23:
hmm, this suggest that issue is in abc directory that will not be removed with git clean -fxd, just feel to execute "rm -rf abc" and then do make again, it will pull abc correctly then.

emard, 2019-08-24:
I will now
rm -rf yosys
git clone https://github.com/YosysHQ/yosys
cd yosys
make config-gcc
make

hop this is clean enough

emard, 2019-08-24:
but even the cleanest possible approach won't work
[  5%] ABC: `` Building binary: abc-5776ad0
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:214: abc-5776ad0] Greška 1
make: *** [Makefile:654: abc/abc-5776ad0] Greška 2

emard, 2019-08-24:
It is possible that I had disk full issue (or near disk full)
which is somehow reported as error and not as "no space on device"
I will build on larger pertition...

emard, 2019-08-24:
yes this is it, near disk full but reported as another error...

closed 2019-08-24 by emard