YosysHQ/yosys #1329: asynchronous flip-flop does not work correctly

promach, 2019-08-26:
## Steps to reproduce the issue

Use [dff.v](https://gist.github.com/promach/ae6d49ebca9b9f209f918622c3b5abe7) or the attached [dff.zip](https://github.com/YosysHQ/yosys/files/3539259/dff.zip)

Run _sby -f dff.sby_

Run _gtkwave dff_cover.gtkw_

## Expected behavior

[enable condition](https://gist.github.com/promach/ae6d49ebca9b9f209f918622c3b5abe7#file-dff-v-L28) should not be triggered.

## Actual behavior

[enable condition](https://gist.github.com/promach/ae6d49ebca9b9f209f918622c3b5abe7#file-dff-v-L28) is triggered instead.

_q_ is [updated](https://i.imgur.com/WGyDuvH.png) with the value of _d_ which is not supposed to happen.

I am not sure whether this is related to [ice40 asynchronous flip-flop issue](https://github.com/YosysHQ/yosys/issues/470#issuecomment-353870675)

daveshah1, 2019-08-27:
I am not seeing the problem here? Infinitesimally before the clock edge, `d` and `enable` are sampled and are 1, therefore the output is set to 1 infinitesimally after the clock edge.

cliffordwolf, 2019-08-27:
> Expected behavior
> enable condition should not be triggered.

I have no idea why that would be the expected behavior. In fact, as @daveshah1 pointed out, the circuit behaves exactly as it is supposed to behave.

Just consider the following simple example that you can simulate with our favorite simulator.

```
module test;
	reg clock;
	always #5 clock = (clock === 1'b0);

	reg reset = 1;
	always @(posedge clock) reset <= 0;

	reg q, d, enable = 1;
        always @(posedge clock) if (!reset) enable <= 0;
        always @(posedge clock) d <= d !== 1'b1;

	always @(posedge clock or posedge reset) begin
		if (reset) begin
			q <= 1'b0;
		end else if (enable) begin
			q <= d;
		end
	end

	initial begin
		$dumpfile("test.vcd");
		$dumpvars(0, test);
		repeat (4) @(posedge clock);
		$finish;
	end
endmodule
```

This produces the following waveform, which matches exactly the waveform I get when running the code you provided.

[image]

(If `q` starts out as `x` or `0` depend on whether the simulator queues `reset = 1` or `always @(posedge clock or posedge reset)` first.)

closed 2019-08-27 by cliffordwolf