YosysHQ/yosys #1331: Add support for SRLC16E

mkurc-ant, 2019-08-27:
Right now there is only support for `SRL16E` which does not provide Q15 output. I'd like to request adding support for `SRLC16E` which has it (as on page 594 of https://www.xilinx.com/support/documentation/sw_manuals/xilinx13_2/7series_scm.pdf).

The version with Q15 output is useful as it can be used as a beginning of a longer SRL chain. I know that the `SRLC32E` can be used instead but in some cases `SRLC16E` is more suitable.

daveshah1, 2019-08-27:
Seems like Vivado accepts SRLC16E (and also uses an inaccessible hardware mode) too, despite it not being mentioned in any of the Vivado docs I can find.

koriakin, 2019-08-27:
Hmmm, that's an interesting case.

On one hand, you are using the wrong document — the version you're using is for schematic entry and contains a lot of macros that aren't natively supported by the hardware, instead being broken down into pieces or emulated in some way.  The correct document to use is https://www.xilinx.com/support/documentation/sw_manuals/xilinx13_2/7series_hdl.pdf , which should contain only actual primitives and does not mention SRLC16E.

On the other hand, I see no trivial way to reduce SRLC16E to either SRL16E or SRLC32E, and instantiating it in ISE seems to use hardware modes not accessible by those.  Perhaps they missed this primitive when compiling the document?

@eddiehung what do you think?  IMO we should add this primitive to the library, in case someone wants to instantiate it.  The question is, is there anything else in _scm worth adding?

mkurc-ant, 2019-08-27:
According to my knowledge: a LUT can be put into SRL32 mode or 2xSRL16 mode.

The SRL32 instantiated via SRLC32E has Q and Q31 outputs which correspond to O6 and MC31 outputs of the LUT respectively.

When using the LUT in 2xSRL16 mode the "first" SRL16 has its Q output connected to O5 of the LUT. It does not have the Q15 output, hence it can only be used via SRL16E. But the "second" one has its Q connected to O6 and Q15 to MC31 which means that it can be used as SRLC16E.

I also checked that indeed Vivado accepts SRLC16E and allows to use the Q15 output.

I'm just asking to add a simulation model for SRLC16E with Q15 output. The way shift registers are mapped to Xilinx technology may remain unchanged.

commit 0e9b9346 (2019-08-27): xilinx: Add SRLC16E primitive.

Fixes #1331.
commit d361f5ab (2019-08-27): xilinx: Add SRLC16E primitive.

Fixes #1331.
PR #1333 (2019-08-27): xilinx: Add SRLC16E primitive.
Fixes #1331.
closed 2019-08-27 by koriakin