YosysHQ/yosys #1345: 'passes/pmgen/xilinx_srl_pm.h' file not found

satnam6502, 2019-08-30:
I downloaded the current Yosys onto my Mac, ran the brew step which worked:

```
satnam-macbookpro3:yosys satnam$ brew tap Homebrew/bundle && brew bundle
Using bison
Using flex
Using gawk
Using libffi
Using git
Using graphviz
Using pkg-config
Using python3
Using tcl-tk
Homebrew Bundle complete! 9 Brewfile dependencies now installed.
```

and then typed `make` which failed with:

```
[ 47%] Building passes/pmgen/test_pmgen.o
passes/pmgen/test_pmgen.cc:31:10: fatal error: 'passes/pmgen/xilinx_srl_pm.h' file not found
#include "passes/pmgen/xilinx_srl_pm.h"
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make: *** [passes/pmgen/test_pmgen.o] Error 1
satnam-macbookpro3:yosys satnam$ find . -name xilinx_srl_pm.h
```

Git log:

```
satnam-macbookpro3:yosys satnam$ git log
commit 999fb33fd0e6d1714bc61f7e50a5c16e0da9d4ab (HEAD -> master, origin/master, origin/HEAD)
Merge: eef06761 6a111ad3
Author: Eddie Hung <eddie@fpgeh.com>
Date:   Fri Aug 30 12:27:09 2019 -0700

    Merge pull request #1340 from YosysHQ/eddie/abc_no_clean

    abc9 to not call "clean" at end of run (often called outside)
```

daveshah1, 2019-08-30:
Looks like Eddie just fixed this one in 17b77fd. Could you pull latest master and try again?

satnam6502, 2019-08-30:
Yup! Fixed! Thank you!

```
satnam-macbookpro3:yosys satnam$ git pull
remote: Enumerating objects: 20, done.
remote: Counting objects: 100% (20/20), done.
remote: Compressing objects: 100% (14/14), done.
remote: Total 20 (delta 8), reused 9 (delta 6), pack-reused 0
Unpacking objects: 100% (20/20), done.
From https://github.com/YosysHQ/yosys
   999fb33f..17b77fd4  master     -> origin/master
   4e782f15..89359b69  xc7dsp     -> origin/xc7dsp
Updating 999fb33f..17b77fd4
Fast-forward
 passes/pmgen/Makefile.inc | 2 +-
 1 file changed, 1 insertion(+), 1 deletion(-)
satnam-macbookpro3:yosys satnam$ make
$BREW_PREFIX is [/usr/local/opt]
[  0%] Building kernel/version_17b77fd4.cc
[  0%] Building kernel/version_17b77fd4.o
[  0%] Building passes/pmgen/xilinx_srl_pm.h
[ 47%] Building passes/pmgen/test_pmgen.o
[ 48%] Building passes/pmgen/ice40_dsp.o
```

closed 2019-08-30 by satnam6502