YosysHQ/yosys #1353: Can't create IdString objects in pyosys

thasti, 2019-09-04:
For example when getting the connection for a cell port `\Q`, one would access the `connections_` dictionary of the cell using an IdString. In the past I did this as follows:

```
conn = cell.connections_[ys.IdString("\Q")]
```

With current master (Yosys 0.9+36, git sha1 58af64b6) the creation of the IdString obect leads to an error:

```
>>> ys.IdString("\Q")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
Boost.Python.ArgumentError: Python argument types in
    None.__init__(IdString, str)
did not match C++ signature:
    __init__(_object*)
```

It seems IdString objects can only be created from Python without supplying a string.

Is this expected behavior? Is there a different way to accomplish the functionality outlined in the beginning instead?

daveshah1, 2019-09-11:
I think the cause of this is https://github.com/YosysHQ/yosys/commit/8222c5735e3b31fc2a7507abe607bb5aa9f863de making the IdString constructors `inline`, which the automagical Python wrapper generator isn't able to parse.

However, the original design intention of the Python API was to not need IdString construction at all but to automatically convert between Python strings and IdStrings, as is done in nextpnr. I don't think this ever worked though, so isn't a regression.

cc @btut

thasti, 2019-09-11:
Agreed, looks like that is the case. Indeed this never worked (for me at least), which is why I resorted to constructing IdStrings in the first place. It is still a regression for me as IdStrings are exposed via the API and their behavior changed between releases. Anything that could be done to help here?

daveshah1, 2019-09-11:
The solution for now would be to fix the wrapper generator to handle `inline` constructors

cliffordwolf, 2019-09-20:
ping @btut @christian-krieg

PR #1397 (2019-09-30): Generate Python wrappers for inline constructors
Fixes: #1353
commit f3926980 (2019-09-30): Generate Python wrappers for inline constructors

Fixes: #1353
closed 2019-09-30 by eddiehung