YosysHQ/yosys #1357: gzip_verilog.v.gz failing on macOS

emilazy, 2019-09-06:
## Steps to reproduce the issue

As part of https://github.com/NixOS/nixpkgs/pull/68107, I'm trying to get the tests to pass on macOS when built with Nix. Unfortunately, the gzip_verilog test [seems to be failing](https://dev.azure.com/arcnmx/nixexprs/_build/results?buildId=970) and I'm not sure why. Perhaps it's failing to link to `zlib`? But it's included in the build dependencies so there should be no problems there. Unfortunately I don't have direct access to a macOS machine so I can't tell whether this is a general macOS problem or something happening due to the Nix build environment. Any suggestions on how to narrow down this failure would be appreciated.

## Expected behavior

All tests pass, like they do on Linux.

## Actual behavior

```
Running gzip_verilog.ys..
gzip_verilog.v.gz:1: ERROR: syntax error, unexpected $undefined
make[1]: *** [run-test.mk:25: run-gzip_verilog.ys] Error 1
```

daveshah1, 2019-09-11:
It seems unlikely that failing to link zlib is the problem here (there is an option to not build the zlib support, but that should print an error if gzip magic is found).

The most likely options are either Yosys is failing to detect the gzip magic in the first place due to some kind of `istream` issue, or corruption is happening during the decompression.

The tests are run in quiet mode, `-q`, afaik. But with this disabled, it should print "Found gzip magic in file `%s', decompressing using zlib.\n" if the gzip magic detection is working.

Then it would probably be worth adding something to dump the decompressed output.

Unfortunately I've got no access to macOS either. But I think @mmicko might.

mmicko, 2019-09-23:
Will check on macOS with native build, may have some questions for  nix builds

mmicko, 2019-09-23:
Found source for this error. But also there is issue creating gzip output that is separate issue, that I as well need to handle.

closed 2019-09-27 by cliffordwolf

emilazy, 2019-09-27:
Thank you @mmicko for fixing this!

commit 435300f9 (2019-09-27): Make read/write gzip files on macos works, fixes #1357
commit 7bde5554 (2019-09-27): Merge pull request #1404 from YosysHQ/fix_gzip_macos

Make read/write gzip files on macos works, fixes #1357