YosysHQ/yosys #1361: Assert failure at startup using x86_64-w64-mingw32-g++ on Ubuntu 18.04

xobs, 2019-09-08:
Yosys silently fails an assert and exits without printing anything when built using `x86_64-w64-mingw32-g++` on Ubuntu 18.04.

When built with debug symbols and run under `gdb.exe`, I can place a breakpoint on `_exit` and get the following backtrace:

```
Breakpoint 2, 0x00000000007c9ec8 in _exit ()
(gdb) bt
#0  0x00000000007c9ec8 in _exit ()
#1  0x000000000043ae32 in Yosys::logv_error_with_prefix(char const*, char const*, char*) () at /usr/share/mingw-w64/include/process.h:46
#2  0x000000000043ae49 in Yosys::logv_error(char const*, char*) () at kernel/log.cc:343
#3  0x000000000043ae6c in Yosys::log_error(char const*, ...) () at kernel/log.cc:392
#4  0x0000000000404de9 in Yosys::log_assert_worker () at ./kernel/log.h:153
#5  0x00000000007e37fd in Yosys::RTLIL::IdString::get_reference(char const*) () at ./kernel/rtlil.h:148
#6  0x00000000007e4520 in _ZN5Yosys5RTLIL8IdStringC4EPKc () at ./kernel/rtlil.h:258
#7  0x00000000008fdbdf in _GLOBAL__sub_I_opt_share.cc () at ./kernel/rtlil.h:258
#8  0x00000000007ba072 in __do_global_ctors ()
#9  0x00000000004013cb in __tmainCRTStartup ()
#10 0x000000000040151b in mainCRTStartup ()
(gdb)
```

Removing the assert fixes the issue, however that's obviously not a good solution.

There are two issues here:

1. The assert is failing, and I don't know why.
2. The program is silently exiting without printing anything, probably because it's exiting before the console output has been set up.

xobs, 2019-09-08:
Further information:

This is being built with `make config-msys2-64` using the toolchain available in Ubuntu 18.04.  I am testing this on a Windows 10 machine.

Things I've tried so far to fix it, in no particular order:

* Building with both x86_64-w64-mingw32-g++-posix and x86_64-w64-mingw32-g++-win32
* Building with `-std=c++17`
* Building with `-mthreads`
* Building both stripped and unstripped binaries

Of note is that gdb appears unable to actually set breakpoints, which makes this a challenge to debug:

```
(gdb) b kernel/log.h:153
No source file named kernel/log.h.
Make breakpoint pending on future shared library load? (y or [n]) n
(gdb) b Yosys::log_assert_worker()
Can't find member of namespace, class, struct, or union named "Yosys::log_assert_worker"
Hint: try 'Yosys::log_assert_worker()<TAB> or 'Yosys::log_assert_worker()<ESC-?>
(Note leading single quote.)
Make breakpoint pending on future shared library load? (y or [n]) n
(gdb)
```

xobs, 2019-09-08:
Since I can't put a breakpoint on that value, I modified it to the following:

```cpp
                static int get_reference(const char *p)
                {
                        //log_assert(destruct_guard.ok);
                        if (!destruct_guard.ok)
                                guard_failures++;
                        else
                                guard_successes++;
```

Then I ran yosys and exited right away:

```
Breakpoint 1, 0x00000000007d1d50 in _exit ()
(gdb) p guard_failures
$1 = 2
(gdb) p guard_successes
$2 = 198
(gdb)
```

Reading from https://isocpp.org/wiki/faq/ctors#static-init-order it seems as though this is undefined behavior in C++, since some static RTLIL object is being constructed at startup that contains a static call to a static `destruct_guard_t`, which may not yet have been constructed.

I'm inclined to believe that the correct solution is to disable the assert, since it seems the result is undefined anyway.  However, I would like someone who understands C++ and Yosys better to comment on the situation.

xobs, 2019-09-08:
The assert seems to be showing that there are other issues.  When I try to synthesize a design using this version of yosys, it fails elsewhere:

```
2.25.1. Executing Verilog-2005 frontend: D:\Software\Icestorm\bin\../share/yosys/ice40/brams_map.v
Parsing Verilog input from `D:\Software\Icestorm\bin\../share/yosys/ice40/brams_map.v' to AST representation.
Generating RTLIL representation for module `\$__ICE40_RAM4K'.
Generating RTLIL representation for module `\$__ICE40_RAM4K_M0'.
Generating RTLIL representation for module `\$__ICE40_RAM4K_M123'.
D:\Software\Icestorm\bin\../share/yosys/ice40/brams_map.v:277: ERROR: Parameter _TECHMAP_REPLACE_.INIT_0 with non-constant value!
```

More things I've tried:

* Compiling with `-O0`
* Replacing the compiler with `gcc` (i.e. to ensure it's a problem with `x86_64-w64-mingw32-g++`) which produces a working binary
* Running under `wineconsole` (`WINEDEBUG=+relay wineconsole --backend=curses ./yosys.exe 2>&1 | grep ERROR` shows the assert message as a call to `msvcrt.strlen`)

So there definitely is something weird going on, I'm just not sure what it is.  Or what "RTLIL" is.

xobs, 2019-09-09:
I've done a bisect, and https://github.com/YosysHQ/yosys/commit/3d3779b0376b8204ed7637053176a07b7271ac1d is the commit that's causing issues.  It works fine with https://github.com/YosysHQ/yosys/commit/a29814ca3f448481d351bc9f8ba0a20084afd883, but not on the following commit.

eddiehung, 2019-09-09:
Thanks for bisecting. I was going to suggest that https://github.com/YosysHQ/yosys/pull/1275 was the root cause, since it uses some exotic incantation to try and improve performance of IdStrings. 

Commit https://github.com/YosysHQ/yosys/commit/3d3779b0376b8204ed7637053176a07b7271ac1d should simply apply that optimisation to more things across the codebase.
A workaround would be to not apply this opt to mingw?

xobs, 2019-09-09:
I have a fix for it, and I'm kind of surprised this isn't affecting other platforms.  I think it's caused by UB in C++.

xobs, 2019-09-09:
I can confirm that 8d128ba6d079fd5f0741c31a9308bf06aaf4673c fixes the issue.

commit 8d128ba6 (2019-09-11): passes: opt_share: don't statically initialize mergeable_type_map

In 3d3779b0376b8204ed7637053176a07b7271ac1d this got turned from a
`std::map<std::string, std::string>` to `std::map<IdString, IdString>`.
Consequently, this exposed some initialization sequencing issues (#1361).

Only initialize the map when it's first used, to avoid these static issues.

This fixes #1361.
closed 2019-09-11 by daveshah1