YosysHQ/yosys #1381: techmap-ing module with underspecified input port connections creates new undriven signal

eddiehung, 2019-09-16:
## Steps to reproduce the issue

```
read_verilog <<EOT
module sub(input i, output o, input j);
foobar f(i, o, j);
endmodule
EOT
design -stash techmap

read_verilog <<EOT
(* blackbox *)
module sub(input i, output o, input j);
endmodule

(* blackbox *)
module foobar(input i, output o, input j);
endmodule

module top(input i, output o);
sub s0(i, o);
endmodule
EOT

### techmap -map %techmap
hierarchy -check
check -assert
```
Remove `###` to error out on `check -assert`

## Expected behavior

Identical behaviour as with `techmap` commented out -- no error.

## Actual behavior

```
5. Executing CHECK pass (checking for obvious problems).
checking module top..
Warning: Wire top.\s0.j is used but has no driver.
found and reported 1 problems.
```
where resulting netlist is:
```
module \top
  attribute \src "<<EOT:9"
  wire input 1 \i
  attribute \src "<<EOT:9"
  wire output 2 \o
  attribute \src "<<EOT:10|<<EOT:1"
  wire \s0.i
  attribute \src "<<EOT:10|<<EOT:1"
  wire \s0.j
  attribute \src "<<EOT:10|<<EOT:1"
  wire \s0.o
  attribute \module_not_derived 1
  attribute \src "<<EOT:10|<<EOT:2"
  cell \foobar \s0.f
    connect \i \i
    connect \j \s0.j
    connect \o \o
  end
end
```

eddiehung, 2019-09-16:
All I want to do is to swap a cell's type, preserving everything else...

eddiehung, 2019-09-16:
A fix could be as simple as clearing out any port connections that do not exist on the original cell:

```
                        for (auto &it2 : c->connections_) {
-                               apply_prefix(cell->name, it2.second, module);
-                               port_signal_map.apply(it2.second);
+                               if (cell->connections_.count(it2.first)) {
+                                       apply_prefix(cell->name, it2.second, module);
+                                       port_signal_map.apply(it2.second);
+                               }
+                               else
+                                       it2.second = SigSpec();
```
which fixes the above test case.

Question is -- is this the right/better thing to do?

cliffordwolf, 2019-09-18:
> A fix could be as simple as clearing out any port connections that do not exist on the original cell:  
> ...

That doesn't look correct. Consider this test case:

```
read_verilog <<EOT
module sub(input a, input b, output c);
subsub s1(.b(a), .y(c));
endmodule

module subsub(input b, output y);
assign y = b;
endmodule

module top(input i, output o);
sub s0(.a(i), .c(o));
endmodule
EOT

prep -flatten -top top
opt_clean -purge
dump
```

Obviously this should produce a result that has `o = i`, which it does with Yosys master, but with your patch `o` and `i` end up not connected.

cliffordwolf, 2019-09-18:
> All I want to do is to swap a cell's type, preserving everything else...

Jfyi, the `chtype` command does that without the overhead of `techmap`.

eddiehung, 2019-09-18:
> Jfyi, the `chtype` command does that without the overhead of `techmap`.

That's just the boiled down motivation, and all you get when you insist on MVCEs! The original techmap pass where this is from does more.

eddiehung, 2019-09-18:
Before I invest more time into the solution, is this (blanking out connections) the right thing to do? We discussed how `hierarchy` was supposed to fill in those ports to begin with, such that my original test case (without `techmap`) would also fail `check -assert` which would void this issue.

cliffordwolf, 2019-09-19:
I think #1384 fixes this issue. The following change is required in the test script:

```
module sub(input i, output o, (* techmap_autopurge *) input j);
foobar f(i, o, j);
endmodule
```

commit b76fac3a (2019-09-19): Add techmap_autopurge attribute, fixes #1381
PR #1384 (2019-09-20): Add techmap_autopurge attribute
Fixes #1381
closed 2019-09-20 by cliffordwolf