YosysHQ/yosys #1382: Accessing large registers results in excessive compile-time burden (read_verilog)

hzeller, 2019-09-17:
## Steps to reproduce the issue

Simplified example:

```bash
cat > /tmp/large_register.v <<EOF
module large_register(input  	   clk,
		      input 	   valid,
		      input [19:0] addr,
		      input 	   wr_bit
		      );
   reg [1<<20:0] large_register;
   always @(posedge clk) begin
      large_register[addr] <= (valid ? wr_bit : large_register[addr]);
   end
endmodule
EOF

# Now running yosys on this will eat a lot of CPU and memory
yosys -QTp "read_verilog /tmp/large_register.v"
```
Using version at head: 36df37a734c3dd315f00edbe19b364b6f9a1a132

## Expected behavior

Even though the register in question is somewhat large, the expectation is be that the code would be processed with reasonable use of CPU and memory.

## Actual behavior

Yosys uses 10G+ memory and uses 100% CPU for > 30min (interrupted after that) to just `read_verilog` this code.

hzeller, 2019-09-18:
Using the Linux tools `perf record` and `perf report` to just looking at the CPU time, it seems that a lot of happens due to representation as `SigBit` with CPU use in inserting/lookup-ing these structures.

(struct SigBit uses at least 12 bytes)

```
  21.03% [.] Yosys::hashlib::dict<Yosys::RTLIL::SigBit, Yosys::RTLIL::SigBit, Yosys::hashlib::hash_ops<Yosys::RTLIL::SigBit> >::find
  16.84% [.] Yosys::RTLIL::SigSpec::operator=
  16.43% [.] Yosys::hashlib::dict<Yosys::RTLIL::SigBit, Yosys::RTLIL::SigBit, Yosys::hashlib::hash_ops<Yosys::RTLIL::SigBit> >::do_lookup
  15.63% [.] std::vector<Yosys::RTLIL::SigBit, std::allocator<Yosys::RTLIL::SigBit> >::emplace_back<Yosys::RTLIL::SigBit>
  11.42% [.] Yosys::RTLIL::SigSpec::unpack
  11.24% [.] Yosys::RTLIL::SigSpec::replace
  ...
```

whitequark, 2019-11-15:
> Even though the register in question is somewhat large, the expectation is be that the code would be processed with reasonable use of CPU and memory.

The register you're using here is only 16 times larger than the minimum vector length required by IEEE 1364-2005, so I would say this is indeed a bug in Yosys.

[image]

zachjs, 2021-03-04:
I tried to replicate this slowdown using the latest, but the frontend only took a couple seconds on this example. Running without the fix in #1414 only made it marginally (~10%) slower. I'm wondering if this is still an issue, or if there are other factors at play.

hzeller, 2021-03-04:
The #1414 was only addressing one part of the problem, IIRC.
I can confirm that this indeed seems to be fixed in the meantime, so looks like this can be closed.

I can try to do a bisecting later to find which change actually fixed it, but feel free to close this now.

closed 2021-03-04 by whitequark