YosysHQ/yosys #1390: \TRANSPARENT=1 prevents BRAM inference if $memrd and $memwr are in asynchronous domains

whitequark, 2019-09-20:
## Steps to reproduce the issue

Use [x.zip](https://github.com/YosysHQ/yosys/files/3637380/x.zip). Run `yosys x.il -p 'synth_ice40 -run :map_ffram'`.

## Expected behavior

A BRAM is inferred, and the `\TRANSPARENT` parameter is ignored. As @cliffordwolf said elsewhere:
> If there are two (non-synchronized) clock domains then TRANSPARENT is only relevant for write ports that are in the same clock domain as the transparent read port.

## Actual behavior

No BRAM is inferred:
```
    Mapping to bram type $__ICE40_RAM4K_M0 (variant 1):
      Write port #0 is in clock domain \write_clk.
        Mapped to bram port B1.
      Read port #0 is in clock domain \read_clk.
        Bram port A1.1 cannot have soft transparency logic added as read and write clock domains differ.
        Failed to map read port #0.
    Mapping to bram type $__ICE40_RAM4K_M0 failed.
```

daveshah1, 2019-09-20:
Not sure if I have a record of a discussion anywhere, but at the time I made this change (previously it tried to create bypass logic which usually made things much worse), I was concerned about synth/sim mismatches - if the two clocks happen to be the same there will be defined behaviour and a mismatch.

whitequark, 2019-09-20:
That makes sense, but makes it impossible to express some true dual-port RAM configurations. Consider Xilinx TDP RAM, port A in WRITE_FIRST mode, with port A as read/write and port B as read-only. There would be no way to express this then.

daveshah1, 2019-09-20:
It seems to me that a better solution would be a way of specifying which read ports are transparent wrt write ports, rather than using clock domains as a proxy for this - even if it would mean more significant changes to the way ports are specified.

whitequark, 2019-09-20:
I agree, this is actually what I wanted to do in nMigen in https://github.com/m-labs/nmigen/issues/216.

cliffordwolf, 2019-09-21:
Quick note: For inference Yosys has no way of knowing if the two clocks are synchronized or not.

whitequark, 2019-09-21:
Of course. I think @daveshah1's solution is a good one, but I don't really know how to express it in RTLIL. Maybe add something like `\DOMAIN=[int]` and all ports with the same `\DOMAIN` would be considered synchronized for synthesis?

daveshah1, 2019-09-21:
I don't think clock synchronisation is even the only problem here. Consider the following verilog, roughly similar to some Xilinx examples so would be good to support eventually

```verilog

wire [7:0] addra, addrb, wdata;
wire wren;
reg [7:0] rdataa, rdatab;

reg [7:0] mem[0:255]

always @(posedge clk) begin
    if (wren) begin
        mem[addra] <= wdata;
        rdataa <= wdata;
   end else
        rdataa <= mem[addra];

always @(posedge clk)
    rdatab <= mem[addrb];
end

```

Despite being in the same clock domain `rdataa` is transparent whereas `rdatab` isn't.

whitequark, 2019-09-21:
So `rdataa` would have `\DOMAIN`=0, `\TRANSPARENT`=1, and `rdatab` would have `\DOMAIN`=0, `\TRANSPARENT`=0.

whitequark, 2019-09-21:
Oh, I see, that won't work.

daveshah1, 2019-09-21:
You are right, the example I was actually thinking of was a TDP one:

```verilog
wire [7:0] addra, addrb, wdataa, wdatab;
wire wrena, wrenb;
reg [7:0] rdataa, rdatab;

reg [7:0] mem[0:255]

always @(posedge clk) begin
    if (wrena) begin
        mem[addra] <= wdataa;
        rdataa <= wdataa;
   end else
        rdataa <= mem[addra];
end

always @(posedge clk) begin
    if (wrenb) begin
        mem[addrb] <= wdatab;
        rdatab <= wdatab;
   end else
        rdatab <= mem[addrb];
end
```

This is distinguished from a case where the read ports are transparent to all write ports (which is the transparent sematics that Yosys generally accepts at the moment):
```verilog
wire [7:0] addra, addrb, wdataa, wdatab;
wire wrena, wrenb;
reg [7:0] addrla, addrlb;
wire [7:0] rdataa, rdatab;

reg [7:0] mem[0:255]

always @(posedge clk) begin
    if (wrena) begin
        mem[addra] <= wdataa;
   end
   addrla <= addra;
end
assign rdataa = mem[addrla];

always @(posedge clk) begin
    if (wrenb) begin
        mem[addrb] <= wdatab;
   end
   addrlb <= addrb;
end
assign rdatab = mem[addrlb];
```
The one case I know where behaviour is defined (because only a single clock for both ports is allowed) is Xilinx URAM. In that case, one port always performs its action after the other (I can't remember the exact ordering), same-port collisions effectively implement NO_CHANGE mode but I guess bypass muxes could be added in synthesis.