YosysHQ/yosys #1407: in MXE build abc9 crashes on read <abc-temp-dir>/input.xaig

Chopper455, 2019-09-27:
## Steps to reproduce the issue
Compiled 143f82d with 32-bit and 64-bit MXE, run in Win 7.
Used simple 4-bit counter:

[counter4.zip](https://github.com/YosysHQ/yosys/files/3661661/counter4.zip)

As a target used xc7 with commands:
>read_verilog counter4.v
synth_xilinx -family xc7 -abc9 -top counter4 -edif proj_xilinx.edf

## Expected behavior

In xubuntu 18.04 with 64-bit build block compiles successfully for the same 143f82d tag.
>2.39.1.4. Executing ABC9.
Running ABC command: <yosys-exe-dir>/yosys-abc -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
ABC: 
ABC: + read_lut <yosys-exe-dir>/share/xilinx/abc_xc7.lut 
ABC: + read_box -v <yosys-exe-dir>/share/xilinx/abc_xc7.box 
ABC: + &read <abc-temp-dir>/input.xaig 
ABC: + &ps 
ABC: <abc-temp-dir>/input : i/o =      6/     10  and =       0  lev =    0 (0.00)  mem = 0.00 MB  box = 1  bb = 0
ABC: + &st 
ABC: + &scorr 
ABC: Warning: The network is combinational.
ABC: + &sweep 
ABC: + &dc2 
ABC: + &st 
ABC: + &dch -f 
ABC: + &ps 
ABC: <abc-temp-dir>/input : i/o =      6/     10  and =       0  lev =    0 (0.00)  mem = 0.00 MB  ch =    0  box = 1  bb = 0
ABC: + &if -W 300 -v 
ABC: K = 8. Memory (bytes): Truth =    0. Cut =   64. Obj =  144. Set =  672. CutMin = no
ABC: Node =       0.  Ch =     0.  Total mem =    0.01 MB. Peak cut mem =    0.00 MB.
ABC: P:  Del =  921.00.  Ar =       0.0.  Edge =        0.  Cut =        0.  T =     0.00 sec
ABC: P:  Del =  921.00.  Ar =       0.0.  Edge =        0.  Cut =        0.  T =     0.00 sec
ABC: P:  Del =  921.00.  Ar =       0.0.  Edge =        0.  Cut =        0.  T =     0.00 sec
ABC: F:  Del =  921.00.  Ar =       0.0.  Edge =        0.  Cut =        0.  T =     0.00 sec
ABC: A:  Del =  921.00.  Ar =       0.0.  Edge =        0.  Cut =        0.  T =     0.00 sec
ABC: A:  Del =  921.00.  Ar =       0.0.  Edge =        0.  Cut =        0.  T =     0.00 sec
ABC: Total time =     0.00 sec
ABC: + &mfs 
ABC: Error: Abc_CommandAbc9Mfs(): The current white-boxes have more than 6 inputs. Cannot use "mfs".
ABC: + &ps -l 
ABC: <abc-temp-dir>/input : i/o =      6/     10  and =       0  lev =    0 (0.00)  mem = 0.00 MB  box = 1  bb = 0
ABC: Mapping (K=0)  :  lut =      0  edge =       0  lev =    0 (0.00)  levB =    1  mem = 0.00 MB
ABC: LUT = 0 : Ave = 0.00
ABC: + &write <abc-temp-dir>/output.aig 
<suppressed ~41 debug messages>
Removed 0 unused cells and 30 unused wires.

## Actual behavior

I've compiled tags 143f82d 0a2d8db and 903cd58 with both 32-bit and 64-bit MXE builds in xubuntu 18.04. In 64-bit build ABC was without pthread support.
All of them cause ABC to exit on assertion:

>2.39.1.4. Executing ABC9.
Running ABC command: <yosys-exe-dir>/yosys-abc -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
ABC: 
ABC: + read_lut <yosys-exe-dir>/share\xilinx/abc_xc7.lut 
ABC: + read_box -v <yosys-exe-dir>/share\xilinx/abc_xc7.box 
ABC: + &read <abc-temp-dir>/input.xaig 
ABC: 
ABC: This application has requested the Runtime to terminate it in an unusual way.
ABC: Please contact the application's support team for more information.
ABC: Assertion failed!
ABC: 
ABC: Program: <yosys-exe-dir>/yosys-abc.exe
ABC: File: ./src/misc/vec/vecStr.h, Line 356
ABC: 
ABC: Expression: i >= 0 && i < p->nSize_

input.xaig is included
[input.zip](https://github.com/YosysHQ/yosys/files/3661632/input.zip)

mmicko, 2019-09-27:
Was able to reproduce this, thanks for example.

eddiehung, 2019-09-27:
The default looks like it's coming from ABC. I'll check to see if the XAIG runs through a Linux build of ABC in about an hour..

mmicko, 2019-09-27:
have confirmed that linux build works fine, so it is something windows related

eddiehung, 2019-09-27:
> have confirmed that linux build works fine, so it is something windows related

Linux Yosys+ABC build, or Linux ABC build using same XAIG? My concern is around whether Yosys is generating an invalid XAIG on Windows.

eddiehung, 2019-09-27:
Okay, this is definitely a Yosys issue, since the same input.xaig (generated on Windows) fails on Linux abc:

```
$ ./yosys-abc -c "&r input.xaig"
ABC command line: "&r input.xaig".

yosys-abc: ./src/misc/vec/vecStr.h:356: char Vec_StrEntry(Vec_Str_t *, int): Assertion `i >= 0 && i < p->nSize' failed.
```

mmicko, 2019-09-27:
Binary compare also shows \n (on linux) and \r\n (on windows) being used on some places, so will see if it is up to that

eddiehung, 2019-09-27:
Ooh. Good thinking. I'm starting to dig and see that Windows' input.xaig is 540 bytes and Linux is 472...

eddiehung, 2019-09-27:
Looks like that was it. I did a "binary sed" on the input.xaig:
`bbe -e "s/\r\n/\n/g" input.xaig > input2.xaig`
and then fed it to ABC, which seemed to work:
```
 ./yosys-abc -c "&r input2.xaig"
ABC command line: "&r input2.xaig".

Warning: Creating unit-delay box delay tables because box library is not available.
Warning: Creating unit-delay box delay tables because box library is not available.
```

eddiehung, 2019-09-27:
Just for completeness, I think we have an idea of where it breaks. Basically, the `ostream` isn't opened in binary mode, so Windows will use `\r\n` as its newline combo. A fix is incoming.

xobs, 2019-09-29:
The patch doesn't seem to fix the issue for me.

I'm running 0c380f085576c2cead5e3576825cb60046bfd76b and I still get an assert error:

```
3.28.1.4. Executing ABC9.
Running ABC command: <yosys-exe-dir>/yosys-abc -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
ABC:
ABC: + read_lut <yosys-exe-dir>/../share/yosys/ecp5/abc_5g.lut
ABC: + read_box -v <yosys-exe-dir>/../share/yosys/ecp5/abc_5g.box
ABC: + &read <abc-temp-dir>/input.xaig
ABC: Assertion failed!
ABC:
ABC: Program: <yosys-exe-dir>/yosys-abc.exe
ABC: File: ./src/misc/vec/vecInt.h, Line 423
ABC:
ABC: Expression: i >= 0 && i < p->nSize
ERROR: ABC: execution of command "D:\Software\ecp5-toolchain-windows-v1.5\bin\yosys-abc -s -f C:\Users\smcro\AppData\Local\Temp\\yosys-abc-4FwsXy/abc.script 2>&1" failed: return code 3.
```

mmicko, 2019-09-29:
@xobs Can you please send example that fails for you ? Since code that is given in this one works just nice.

xobs, 2019-09-29:
This is the output from my litex build system.
[1407.zip](https://github.com/YosysHQ/yosys/files/3666585/1407.zip)

I'm using this release of yosys: https://github.com/xobs/toolchain-icestorm/releases/tag/v1.33-fomu which is unpatched based on the tag I mentioned.

mmicko, 2019-09-29:
@xobs Strange, with your build there are still \r\n if file  are you sure your build also contains previous commit 01aeea29e672dcb883346e61c6b2c8ca667edc3c ?

Here is build I created and it used the latest master and with it it does not break as in your build.

https://www.dropbox.com/s/5uvnctugluu7bte/yosys-win32-mxebin-0.9%2B431.zip?dl=0

xobs, 2019-09-29:
```
❯ yosys -V
Yosys 0c380f085576c2cead5e3576825cb60046bfd76b (Fomu build) (git sha1 ec6d882, x86_64-w64-mingw32-g++ 7.3-win32 -Os)
❯
```

This build was from just before it was merged into master.  I'm rebuilding using master, but I thought 0c380f085576c2cead5e3576825cb60046bfd76b was the patch that fixed it.

mmicko, 2019-09-29:
Well it should be in otherwise it would not compile, but still it looks like it is actually not used in build for some reason. Will try doing mingw build locally as well, just to check if there is something related to that

mmicko, 2019-09-29:
@xobs Even since output now gets generated with new binaries, there is still a problem (not caused by other file), working on fix now, need just test whole process and see if there are some other cases.

xobs, 2019-09-29:
My freshly-built yosys binary now spins forever, getting stuck at:

```
3.28.1.4. Executing ABC9.
Running ABC command: <yosys-exe-dir>/yosys-abc -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
ABC:
ABC: + read_lut <yosys-exe-dir>/../share/yosys/ecp5/abc_5g.lut
ABC: + read_box -v <yosys-exe-dir>/../share/yosys/ecp5/abc_5g.box
ABC: + &read <abc-temp-dir>/input.xaig
ABC: + &ps
ABC: <abc-temp-dir>/input : i/o =    910/   1087  and =    3953  lev =   49 (2.90)  mem = 0.10 MB  box = 110  bb = 0    ABC: + &st
ABC: + &scorr
ABC: Warning: The network is combinational.
ABC: + &sweep
ABC: + &dc2
ABC: + &st
ABC: + &dch -f
ABC: + &ps
ABC: <abc-temp-dir>/input : i/o =    910/   1087  and =    4789  lev =   23 (2.20)  mem = 0.11 MB  ch =  613  box = 110  bb = 0
ABC: + &if -W 200 -v
ABC: K = 7. Memory (bytes): Truth =    0. Cut =   60. Obj =  140. Set =  636. CutMin = no
ABC: Node =    4789.  Ch =   544.  Total mem =    1.21 MB. Peak cut mem =    0.10 MB.
ABC: P:  Del = 3070.00.  Ar =    4384.0.  Edge =     5626.  Cut =    58990.  T =     0.02 sec
ABC: P:  Del = 3070.00.  Ar =    4394.0.  Edge =     5622.  Cut =    58534.  T =     0.02 sec
ABC: P:  Del = 3070.00.  Ar =    2671.0.  Edge =     5093.  Cut =   114647.  T =     0.04 sec
ABC: F:  Del = 3070.00.  Ar =    1707.0.  Edge =     4319.  Cut =    91074.  T =     0.04 sec
ABC: A:  Del = 3070.00.  Ar =    1616.0.  Edge =     3964.  Cut =    87701.  T =     0.05 sec
ABC: A:  Del = 3070.00.  Ar =    1592.0.  Edge =     3954.  Cut =    86331.  T =     0.05 sec
ABC: Total time =     0.21 sec
ABC: + &mfs
ABC: Error: Abc_CommandAbc9Mfs(): The current mapping has nodes with more than 6 inputs. Cannot use "mfs".
ABC: + &ps -l
ABC: <abc-temp-dir>/input : i/o =    910/   1087  and =    3933  lev =   23 (2.20)  mem = 0.10 MB  box = 110  bb = 0    ABC: Mapping (K=7)  :  lut =   1080  edge =    3954  lev =    8 (1.24)  levB =   18  mem = 0.05 MB
ABC: LUT = 1080 : 2=220 20.4 %  3=263 24.4 %  4=357 33.1 %  5=182 16.9 %  6=19 1.8 %  7=39 3.6 %  Ave = 3.66
ABC: + &write <abc-temp-dir>/output.aig
```

Is that the issue you're seeing?

mmicko, 2019-09-29:
Yes, that is the issue. Tough it just take more time, but then I checked Linux build and it is instant. Have solution for this.

xobs, 2019-09-29:
For reference, the process that generated that output is still running, and hasn't outputted any more text in the last hour -- it's still running abc, and chewing up one processor core.

mmicko, 2019-09-29:
If you could please do local build out of pull request to tryout , it should be fixed now. Remaining task for me is to see if ascii format files work as well with this change

xobs, 2019-09-29:
Sure, I will do that.  I have a build of yosys running right now.

xobs, 2019-09-29:
I can confirm that 3f70c1fd26eb109c2c4d899cce55f24bbf04acc1 does NOT fix it, but 9e55b234b47b01dc396e793b7f31236c9e87c185 does fix it.

mmicko, 2019-09-29:
Thanks @xobs for checking this.

Chopper455, 2019-09-30:
Sorry for being out.
Just now compiled tag d5f0794 in 32-bit mxe and it works fine for me.
> 2.43.1.4. Executing ABC9.
Running ABC command: <yosys-exe-dir>/yosys-abc -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
ABC: 
ABC: + read_lut <yosys-exe-dir>/share\xilinx/abc_xc7.lut 
ABC: + read_box -v <yosys-exe-dir>/share\xilinx/abc_xc7.box 
ABC: + &read <abc-temp-dir>/input.xaig 
ABC: + &ps 
ABC: <abc-temp-dir>/input : i/o =      6/     10  and =       0  lev =    0 (0.00)  mem = 0.00 MB  box = 1  bb = 0
ABC: + &st 
ABC: + &scorr 
ABC: Warning: The network is combinational.
ABC: + &sweep 
ABC: + &dc2 
ABC: + &st 
ABC: + &dch -f 
ABC: + &ps 
ABC: <abc-temp-dir>/input : i/o =      6/     10  and =       0  lev =    0 (0.00)  mem = 0.00 MB  ch =    0  box = 1  bb = 0
ABC: + &if -W 300 -v 
ABC: K = 8. Memory (bytes): Truth =    0. Cut =   64. Obj =  120. Set =  624. CutMin = no
ABC: Node =       0.  Ch =     0.  Total mem =    0.00 MB. Peak cut mem =    0.00 MB.
ABC: P:  Del =  921.00.  Ar =       0.0.  Edge =        0.  Cut =        0.  T =     0.00 sec
ABC: P:  Del =  921.00.  Ar =       0.0.  Edge =        0.  Cut =        0.  T =     0.00 sec
ABC: P:  Del =  921.00.  Ar =       0.0.  Edge =        0.  Cut =        0.  T =     0.00 sec
ABC: F:  Del =  921.00.  Ar =       0.0.  Edge =        0.  Cut =        0.  T =     0.00 sec
ABC: A:  Del =  921.00.  Ar =       0.0.  Edge =        0.  Cut =        0.  T =     0.00 sec
ABC: A:  Del =  921.00.  Ar =       0.0.  Edge =        0.  Cut =        0.  T =     0.00 sec
ABC: Total time =     0.00 sec
ABC: + &mfs 
ABC: Error: Abc_CommandAbc9Mfs(): The current white-boxes have more than 6 inputs. Cannot use "mfs".
ABC: + &ps -l 
ABC: <abc-temp-dir>/input : i/o =      6/     10  and =       0  lev =    0 (0.00)  mem = 0.00 MB  box = 1  bb = 0
ABC: Mapping (K=0)  :  lut =      0  edge =       0  lev =    0 (0.00)  levB =    1  mem = 0.00 MB
ABC: LUT = 0 : Ave = 0.00
ABC: + &write <abc-temp-dir>/output.aig 
<suppressed ~41 debug messages>
Removed 0 unused cells and 30 unused wires.

Chopper455, 2019-09-30:
As a stress test tried to compile MOS6502 from opencores.
>read_verilog ../rtl/ALU.v
read_verilog ../rtl/cpu.v
synth_xilinx -family xc7 -abc9 -top cpu -edif proj_xilinx.edf
[verilog-6502-master.zip](https://github.com/YosysHQ/yosys/files/3669189/verilog-6502-master.zip)

ABC hangs on the step
>ABC: + &write <abc-temp-dir>/output.aig

mmicko, 2019-09-30:
@Chopper455 this should be already handled in https://github.com/YosysHQ/yosys/pull/1416 since it also appeared at example @xobs pointed out. Please do try switching to "mmicko/frontend_binary_in" branch and try then.

Chopper455, 2019-09-30:
I've compiled th branch in 32-bit mxe.
It passes 4-bit counter, but fails to write-out(hangs) on mos6502(github), PID(opencores), wbuart(opencores) and spi_top(opencores) blocks.
Temp dir archieve for cpu block is attached(just in case).
[yosys-abc-8rkLDj.zip](https://github.com/YosysHQ/yosys/files/3669573/yosys-abc-8rkLDj.zip)

A little later I\ll wire results of linux build.

Chopper455, 2019-09-30:
I've compiled the branch with 32-bit MXE and tried it on several blocks(MOS6502, PID, wbuart, spi_top). Everything passes well.
Thank you!

commit d0493925 (2019-09-28): Support binary files for backends, fixes #1407
commit ce0631c3 (2019-09-29): Merge pull request #1413 from YosysHQ/mmicko/backend_binary_out

Support binary files for backends, fixes #1407
closed 2019-09-30 by mmicko