YosysHQ/yosys #1410: Yosys gives judgemental, unprofessional warning messages

xobs, 2019-09-27:
Yosys gives unprofessional warning messages when encountering certain comments.  While these are indicative of poor coding style that is proprietary, a professional tool shouldn't berate the user for their choice of code, and should instead warn the user about nonstandard coding practices.

## Expected Behavior

```verilog
// synopsys translate_off
`include "timescale.v"
// synopsys translate_on
```

`yosys read_verilog i2c_master_bit.v`

```
Warning: i2c_master_bit_ctl.v:137: Start of vendor-specific `translate_off` comment section
For portability, it is recommended to use `ifdef constructs instead!
```

## Observed Behavior

```verilog
// synopsys translate_off
`include "timescale.v"
// synopsys translate_on
```

`yosys read_verilog i2c_master_bit.v`

```
Warning: Found one of those horrible `(synopsys|synthesis) translate_off' comments.
Yosys does support them but it is recommended to use `ifdef constructs instead!
```

whitequark, 2019-09-27:
A professional tool shouldn't use hot comments in a language where the preprocessor exists either, so I guess Synplify isn't one of them.

<hr>

I think showing the file and line is not particularly useful unless the warning will be replicated for every instance of the hot comment, which is of course an option, but likely to be quite annoying.

rezayat2006, 2020-02-08:
Tnx 6

zachjs, 2021-03-28:
Are we still looking to modify the language of this warning? I think something in the spirit of (the revisions to) #2611 may be worthwhile. We could also link to some outside opinion on "translate_off considered harmful".

whitequark, 2021-03-28:
Let's modify it. I would suggest something along the lines of "instead of hot comments, a standard-compliant construct such as `ifdef is recommended".

PR #2709 (2021-03-30): verilog: revise hot comment warnings
The exact wording and structure likely needs tweaking.

Fixes #1410