YosysHQ/yosys #1415: Yosys synth_ecp5 -abc9 fails with assert

Spritetm, 2019-09-29:
To reproduce:
- Checkout ``https://github.com/Spritetm/hadbadge2019_fpgasoc``
- Checkout the abc9_crash branch
- In the soc directory, run the following command
``yosys -p "read -sv -DBADGE_VER=3 -DBADGE_V3=1 soc.v picorv32/picorv32.v simpleuart.v lcdiface.v qpi_cache/qpimem_cache.v  qpi_cache/qpimem_iface.v qpi_cache/simple_mem.v qpi_cache/simple_mem_words.v arbiter.v pcpi_fastmul_dsp.v video/vid_linerenderer.v video/video_mem.v qpi_cache/qpimem_arbiter.v qpi_cache/qpimem_iface_intl.v usb/usb_trans.v usb/usb_ep_buf.v usb/usb_rx_pkt.v usb/usb_phy.v usb/usb.v usb/usb_tx_pkt.v usb/usb_ep_status.v usb/usb_tx_ll.v usb/usb_rx_ll.v usb/usb_crc.v usb/fifo_sync_shift.v usb_soc.v pic/pic_wrapper.v pic/risc16f84_clk2x.v pic/dpram_1kx16.v qpi_cache/qpimem_dma_rdr.v ledctl.v simpleuart_irda.v dsadc.v pll.v top_fpga.v mul_18x18_ecp5.v hdmi/clk_8_250_125_25.v hdmi/fake_differential.v hdmi/hdmi-encoder.v hdmi/tmds_encoder.v hdmi/vga2dvid.v hdmi/vga.v  video/ram_dp_24x2048_ecp5.v rng.v video/vid_palettemem_ecp5.v video/vid_tilemapmem_ecp5.v video/vid_tilemem_ecp5.v; \                            synth_ecp5 -abc9 -top top_fpga -json soc.json -blif soc.blif"``

The result is that Yosys crashes with:
``ERROR: Assert `!w->port_input' failed in passes/techmap/abc9.cc:97.``

This is with a very recent yosys (commit c372e7baf9c48d41ebdbea4486a72e8dfaaddd3d ).

I tried to run a bugpoint session to pinpoint where the issue is, but it seems to take a long time... will add if I get any results from that before I need my CPU power for something else.

eddiehung, 2019-09-29:
Initial investigation seems to indicate that this is caused by a latch in https://github.com/Spritetm/hadbadge2019_fpgasoc/blob/22252d6286705b2f12febaaf8946ab1af7982c2d/soc/video/vid_linerenderer.v#L275-L312 which `synth_ecp5` maps into LUTs, and that `abc9` is struggling to unravel due to combinatorial loop.

It should be more robust though, as it works without `abc9`, but as a workaround can you try and get rid of that latch?

I've managed to repro it more quickly (but not reproduce it separately as a MVCE) by setting `-top vid_linerenderer` instead:
```
osys -p "read -sv -DBADGE_VER=3 -DBADGE_V3=1 soc.v picorv32/picorv32.v simpleuart.v lcdiface.v qpi_cache/qpimem_cache.v qpi_cache/qpimem_iface.v qpi_cache/simple_mem.v qpi_cache/simple_mem_words.v arbiter.v pcpi_fastmul_dsp.v video/vid_linerenderer.v video/video_mem.v qpi_cache/qpimem_arbiter.v qpi_cache/qpimem_iface_intl.v usb/usb_trans.v usb/usb_ep_buf.v usb/usb_rx_pkt.v usb/usb_phy.v usb/usb.v usb/usb_tx_pkt.v usb/usb_ep_status.v usb/usb_tx_ll.v usb/usb_rx_ll.v usb/usb_crc.v usb/fifo_sync_shift.v usb_soc.v pic/pic_wrapper.v pic/risc16f84_clk2x.v pic/dpram_1kx16.v qpi_cache/qpimem_dma_rdr.v ledctl.v simpleuart_irda.v dsadc.v pll.v top_fpga.v mul_18x18_ecp5.v hdmi/clk_8_250_125_25.v hdmi/fake_differential.v hdmi/hdmi-encoder.v hdmi/tmds_encoder.v hdmi/vga2dvid.v hdmi/vga.v video/ram_dp_24x2048_ecp5.v rng.v video/vid_palettemem_ecp5.v video/vid_tilemapmem_ecp5.v video/vid_tilemem_ecp5.v; synth_ecp5 -abc9 -top vid_linerenderer -json soc.json -blif soc.blif"
```

eddiehung, 2019-09-29:
Actually, just after typing this up I can see that `-abc9` fails to techmap latches. Not quite the same segfault, but along the same vein.
```
read_verilog <<EOT
module top(input s, b, output a);
always @*
    if (s) a = b;
endmodule
EOT
synth_ecp5 -abc9
```
produces an empty result, whereas omitting `-abc9` gives 1xLUT4.

Spritetm, 2019-09-30:
Huh, I wasn't aware that code still inferred latches... I thought I had all possible situations taken care of, but maybe I was too quick to decide that wouldn't lead to latches. Thanks for noticing. Just fixed that and rebuild with abc9, and it indeed seems that fixes the issue.

daveshah1, 2019-09-30:
It shouldn't infer a latch. What is happening is that Yosys in earlier stages doesn't see that, so it creates the logic for a latch. Later optimisations (i.e. abc) work out that the latch is always enabled (never used as storage) so it is optimised away, hence I suspect that you wouldn't have been getting combinational loop warnings/errors.

I would say this is a Yosys issue, although arguably it's a deeper problem in terms how behavioural HDL should actually be synthesised....

This is a MCVE that infers a latch where one wouldn't be expected.

```verilog
module top(input [1:0] x, output reg y);

always @*
	if (x == 0)
		y = 1'b1;
	else if (x == 1)
		y = 1'b0;
	else if (x == 2)
		y = 1'b1;
	else if (x == 3)
		y = 1'b0;
endmodule
```

eddiehung, 2019-09-30:
The issue here is that doesn't cover "all" cases that `x` could take, including `'bx` and `'bz` values. The "correct" thing to do would be to end with an `else` (or set a default value for `y` before the `if`). Vivado also infers a latch with your testcase...

Your analysis of `abc` vs `abc9` is very plausible though. `abc` can figure out the latch is always enabled and thus optimises it away. The `abc9` flow dies while trying to get the design into abc...

daveshah1, 2019-11-21:
FYI, this is a minimal example of the bug in abc9:
```verilog
module top(input [1:0] d, input e, output reg [1:0] q);

always @*
	if (e)
		q <= d;

endmodule
```
just run this through `synth_ice40 -abc9`

PR #1542 (2019-12-02): abc9: Fix breaking of SCCs
This correctly handles the case where more than one wire bit is part of an SCC, by checking for this case when rewiring. It also renames the input, rather than the output; to avoid breaking port naming when the output is also a primary output.

Fixes #1487 and fixes #1415; once this is fixed I will be happy making abc9 the default for ECP5 flows - the area reduction is certainly worth it.
closed 2019-12-02 by daveshah1