YosysHQ/yosys #1418: negative numbers are not sign expanded across module ports

mattvenn, 2019-09-30:
## Steps to reproduce the issue

[image]

[neg_demo.zip](https://github.com/YosysHQ/yosys/files/3671537/neg_demo.zip)

## Expected behavior

FPGA should match simulation

## Actual behavior

If a signed module port is mapped to a larger register, the upper bits are all set to 0, so the negative number becomes positive 2's complement.

eddiehung, 2019-12-12:
Wow, I can't believe this wasn't discovered earlier, as it seems pretty serious. 

My understanding of this is that "signedness" of ports isn't kept in the RTLIL; thus there's no way for `hierarchy` (the pass that does the zero-extension):
```
2.3. Analyzing design hierarchy..
Top module:  \instantiate_negs
Used module:     \neg_demo
Removed 0 unused modules.
Warning: Resizing cell port instantiate_negs.neg_demo_1.out from 16 bits to 8 bits.
```
to know whether to zero extend or to sign extend, and so it just does the former.

Let's consult @cliffordwolf on this, but I can think of two solutions off the top of my head: preserve signedness in RTLIL using an attribute, or hierarchy should reach into the AST to determine signedness of ports.

whitequark, 2019-12-12:
> hierarchy should reach into the AST to determine signedness of ports.

I don't think this can work in case of `connect_rpc`, which does not involve an AST in case the frontend emits RTLIL directly.

PR #2491 (2020-12-22): Sign extend port connections where necessary
- Signed cell outputs are sign extended when bound to larger wires
- Signed connections are sign extended when bound to larger cell inputs
- Sign extension is performed in hierarchy and flatten phases
- genrtlil indirects signed constants through signed wires
- Other phases producing RTLIL may need to be updated to preserve signedness information
- Resolves #1418
- Resolves #2265

This change may need further revisions, but I am eager for feedback!
commit 0d8e5d96 (2020-12-22): Sign extend port connections where necessary

- Signed cell outputs are sign extended when bound to larger wires
- Signed connections are sign extended when bound to larger cell inputs
- Sign extension is performed in hierarchy and flatten phases
- genrtlil indirects signed constants through signed wires
- Other phases producing RTLIL may need to be updated to preserve
  signedness information
- Resolves #1418
- Resolves #2265
closed 2020-12-22 by whitequark