YosysHQ/yosys #1427: peepopt performs unsound optimization

whitequark, 2019-10-02:
## Steps to reproduce the issue

Tested on Yosys 0.9+899 (git sha1 da347b9f).

Testcase:
<details>

```
attribute \top 1
module \top
  wire input 5 \clk_if_0__io
  attribute \init 2'00
  wire width 2 \fsm.addr
  wire width 2 \fsm.addr$next
  wire \fsm.next_state
  attribute \init 1'0
  wire \fsm.state
  wire output 1 \led_0__io
  wire output 2 \led_1__io
  wire output 3 \led_2__io
  wire output 4 \led_3__io
  cell $sshl $2
    parameter \A_SIGNED 0
    parameter \A_WIDTH 1
    parameter \B_SIGNED 0
    parameter \B_WIDTH 2
    parameter \Y_WIDTH 4
    connect \A 1'1
    connect \B \fsm.addr
    connect \Y { \led_3__io \led_2__io \led_1__io \led_0__io }
  end
  cell $dff $techmap\fsm.$procdff$39
    parameter \CLK_POLARITY 1'1
    parameter \WIDTH 2
    connect \CLK \clk_if_0__io
    connect \D \fsm.addr$next
    connect \Q \fsm.addr
  end
  cell $dff $techmap\fsm.$procdff$40
    parameter \CLK_POLARITY 1'1
    parameter \WIDTH 1
    connect \CLK \clk_if_0__io
    connect \D \fsm.next_state
    connect \Q \fsm.state
  end
  cell $mux $techmap\fsm.$procmux$25
    parameter \WIDTH 2
    connect \A 2'11
    connect \B \fsm.addr
    connect \S \fsm.state
    connect \Y \fsm.addr$next
  end
  attribute \full_case 1
  cell $mux $techmap\fsm.$procmux$37
    parameter \WIDTH 1
    connect \A 1'0
    connect \B 1'1
    connect \S \fsm.state
    connect \Y \fsm.next_state
  end
end
```
</details>

The testcase has a design that looks like this:
[image]

Namely, it assigns `\fsm.addr` first to 0b00, and then to 0b11, then encodes it as one-hot and displays on LEDs. Running `synth_ice40` on it produces rather interesting results: it lights up `led1`, which is clearly absurd:

[image]

If I comment out the `peepopt` call in `synth_ice40`, it produces the expected optimized design:

[image]

whitequark, 2019-10-02:
Eddie, I believe you wrote this part of peepopt, so I've assigned you. Feel free to unassign if I made a mistake.

eddiehung, 2019-10-02:
Looks like it is mine, sorry. Will look at it when I get up tomorrow!

PR #1432 (2019-10-08): Refactor peepopt_dffmux and be sensitive to \init when trimming
Fixes #1427. 
`equiv_opt synth_ice40` now passes for that testcase.

Issue was that `peepopt_dffmux` was not checking the `\init` value before trimming.

EDIT: Depends on #1431 -- please let that get merged first.
closed 2019-10-08 by eddiehung