YosysHQ/yosys #1447: The xilinx_dsp packing register mechanism changes memory_pass.memory_dff behaviour when flattening

dh73, 2019-10-11:
## Steps to reproduce the issue
Using a modified excerpt of Marlann memory output -> DSP input as example (zip file), execute the following command:
```
synth_xilinx -abc9 -flatten -top testcase
```
_Zip file_:
[testcase.zip](https://github.com/YosysHQ/yosys/files/3719615/testcase.zip)

## Expected behavior
The **memory_pass.memory_dff** merges the ```$dff``` cell to ```$memrd```, so 2 RAMB36E1 can be used when using ```flatten```.

## Actual behavior
A total of  **256 RAM128X1D** are inferred when using ```-flatten``` only, due ```xilinx_dsp``` packing ```$dff``` from memory output that also connects to B input of DSP48E1, _when enables are involved in any way_.

## Technical Details
### When flattening:
Before  executing ```xilinx_dsp``` pass, the following Verilog excerpt generates the processes/cells that are showed in the Image 1
```verilog
   always @(posedge clk) begin
      m_vec <= mem[rbuff0[64:0] + K]; // $memrd
      K <= rbuff0[127:62] + rbuff0[63];
   end
```
_Image 1_:
[image]

After executing ```xilinx_dsp```,  the ```$procff$45``` (which is the FF connected to the output of the ```$memrd``` cell) is used in ffB2:
```
23. Executing XILINX_DSP pass (pack resources into DSPs).
Analysing testcase.$techmap\mult.$mul$testcase.v:41$30.sliceA[2].mul.sliceB[2].mul for Xilinx DSP packing.
preAdd:     --
ffAD:       -- -- --
ffA2:       $procdff$46 -- --
ffA1:       -- -- --
ffB2:       $procdff$45 -- --  <--- Here
ffB1:       -- -- --
ffD:        -- -- --
dsp:        $techmap\mult.$mul$testcase.v:41$30.sliceA[2].mul.sliceB[2].mul
ffM:        -- -- --
postAdd:    $techmap$techmap\mult.$mul$testcase.v:41$30.sliceA[2].mul.$add$/opt/eda/fpga/FOSS/tests/binaries/bin/../share/yosys/mul2dsp.v:201$72
postAddMux: --
ffP:        -- -- --
overflow:   --
  postadder $techmap$techmap\mult.$mul$testcase.v:41$30.sliceA[2].mul.$add$/opt/eda/fpga/FOSS/tests/binaries/bin/../share/yosys/mul2dsp.v:201$72 ($add)
```
Therefore, in memory pass, the ```memory_dff``` is unable to merge the ```$memrd``` output dff cell:
```
29. Executing MEMORY pass.

29.1. Executing MEMORY_DFF pass (merging $dff cells to $memrd and $memwr).
Checking cell `$memwr$\mem$testcase.v:23$24' in module `\testcase': merged $dff to cell.
Checking cell `$memrd$\mem$testcase.v:18$17' in module `\testcase': no (compatible) $dff found.  <- Here
```
This makes ```memory_bram``` pass to use lutram instead of block rams, due the change of read transparency behavior from the rd port of the memory:
_Bram pass_:
```
Write port #0 is in clock domain \clk.
        Mapped to bram port B1.
      Read port #0 is in clock domain !~async~.
        Bram port A1.1 has incompatible clock type.
        Failed to map read port #0.
```
_Lutram pass_:
```
Mapping to bram type $__XILINX_RAM128X1D (variant 1):
      Write port #0 is in clock domain \clk.
        Mapped to bram port B1.
      Read port #0 is in clock domain !~async~.
        Mapped to bram port A1.1.
      Creating $__XILINX_RAM128X1D cell at grid position <0 0 0>: mem.0.0.0
      [...]
```

### When no flattening:
Since ```xilinx_dsp``` does not touch anything outside *mult* block, the ```$45$dff``` is not packed into DSP input port:
_Before running xilinx_dsp_:
[image]
The ```$45$dff``` is not packed by ```xilinx_dsp```:
```
23. Executing XILINX_DSP pass (pack resources into DSPs).
Analysing $paramod\mult_pipe\W=128.$mul$testcase.v:40$29.sliceA[2].mul.sliceB[2].mul for Xilinx DSP packing.
preAdd:     --
ffAD:       -- -- --
ffA2:       -- -- --
ffA1:       -- -- --
ffB2:       -- -- -- <--- Nothing
ffB1:       -- -- --
ffD:        -- -- --
dsp:        $mul$testcase.v:40$29.sliceA[2].mul.sliceB[2].mul
ffM:        -- -- --
postAdd:    $techmap$mul$testcase.v:40$29.sliceA[2].mul.$add$/opt/eda/fpga/FOSS/tests/binaries/bin/../share/yosys/mul2dsp.v:201$71
postAddMux: --
ffP:        -- -- --
overflow:   --
  postadder $techmap$mul$testcase.v:40$29.sliceA[2].mul.$add$/opt/eda/fpga/FOSS/tests/binaries/bin/../share/yosys/mul2dsp.v:201$71 ($add)
```
Memory pass is able to merge ```$45$dff``` cell:
```
29.1. Executing MEMORY_DFF pass (merging $dff cells to $memrd and $memwr).
Checking cell `$memwr$\mem$testcase.v:23$24' in module `\testcase': merged $dff to cell.
Checking cell `$memrd$\mem$testcase.v:18$17' in module `\testcase': merged data $dff to cell.
```
Therefore, BRAMS are used instead of LUTRAMS:
**When not  flattening**:
```
   Number of wires:                556
   Number of wire bits:           7517
   Number of public wires:          20
   Number of public wire bits:    2257
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:               1825
     BUFG                            1
     CARRY4                        128
     DSP48E1                        32
     FDRE                          680
     LUT2                          982
     RAMB36E1                        2
```

**When flattening**:
```
   Number of wire bits:           7612
   Number of public wires:          16
   Number of public wire bits:    1745
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:               2081
     BUFG                            1
     CARRY4                        128
     DSP48E1                        32
     FDRE                          552
     LUT2                          984
     LUT3                          128
     RAM128X1D                     256
```
Although the issue looks similar to #710 , I think the solution here is to improve ```xilinx_dsp``` register packing.

dh73, 2019-10-14:
The ```xilinx_dsp``` can be called under ```coarse``` synthesis process, since the ```map_dsp``` pass have already selected and processed the DSP cells. This may need a discussion, but moving the ```run("xilinx_dsp");``` process in ```map_dsp``` to: 
```c
if (check_label("coarse")) {
                        run("alumacc");
                        run("share");
                        run("opt");
                        run("fsm");
                        run("opt -fast");
                        run("memory -nomap");
                        run("xilinx_dsp");  // <- Moved here
                        run("opt_clean");
                }
```
Produces the following results:
```bash
=== testcase ===

   Number of wires:               1388
   Number of wire bits:          10623
   Number of public wires:          18
   Number of public wire bits:    2001
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:               3499
     BUFG                            1
     CARRY4                        342
     DSP48E1                        32
     FDRE                          552
     LUT2                         2570
     RAMB36E1                        2
```
So using this way as a WA for now, until we decide what is the best way for the pass to work.

daveshah1, 2019-10-14:
At least when I last looked at it; `xilinx_dsp` was working with `$add`, `$sub`, etc in order to map MACs; so it needed to be run before `alumacc` processes these cells. Some more reordering of passes would be needed to get this to work

dh73, 2019-10-14:
Nina also suggested to use ```memory_dff``` before ```xilinx_dsp```, so we preserve the flow at expense of running that pass twice.

commit 3b405d98 (2019-10-22): Call memory_dff before DSP mapping to reserve registers (fixes #1447)
commit a3a7bb9b (2019-10-22): Merge pull request #1452 from nakengelhardt/fix_dsp_mem_reg

Call memory_dff before DSP mapping to reserve registers (fixes #1447)
closed 2019-10-22 by cliffordwolf