YosysHQ/yosys #1453: Verilog reader doesn't resolve scope correctly when new wire is declared with same named wire in assignment

tomverbeure, 2019-10-17:
# Test1

## Repro Instructions

    yosys -p "read_verilog test1.v"

test1.v
```
module test1 (data);

input wire [127:0] data;

generate if (1)
  begin: g0
        wire [31:0] data = data[63:32];
  end
endgenerate

endmodule
```

## Actual Behavior

```
-- Running command `read_verilog -sv test1.v' --

1. Executing Verilog-2005 frontend: test1.v
Parsing Verilog input from `test1.v' to AST representation.
Generating RTLIL representation for module `\test1'.
test1.v:7: Warning: Range select [63:32] out of bounds on signal `\g0.data': Setting all 32 result bits to undef.
Successfully finished Verilog frontend.

Warnings: 1 unique messages, 1 total
End of script. Logfile hash: c42ece85ad
CPU: user 0.01s system 0.00s, MEM: 42.18 MB total, 13.00 MB resident
Yosys 0.9+932 (git sha1 0d037bf, gcc 5.4.0-6ubuntu1~16.04.11 -Og -fPIC)
```

## Expected Behavior

No warning about an out of bounds signal select.

Commercial tools, as well as iverilog, are doing this right. Verilator with --lint-only complains with a similar error:

```
vagrant@ubuntu-xenial:~/projects/bitec_dp$ /opt/verilator/bin/verilator --lint-only test1.v
%Warning-SELRANGE: test1.v:7: Selection index out of range: 63:32 outside 31:0
                   ... Use "/* verilator lint_off SELRANGE */" and lint_on around source to disable this message.
%Error: test1.v:7: Wire inputs its own output, creating circular logic (wire x=x)
%Error: Exiting due to 1 error(s), 1 warning(s)
%Error: Command Failed /opt/verilator/bin/verilator_bin --lint-only test1.v
```

# Test 2

## Repro Instructions

    yosys -p "read_verilog -sv test2.v" 

test2.v:
```
module test2 (data);

input wire [127:0] data;

generate if (1)
  begin: g0
        wire [31:0] data = data[63:32];

        if (1) begin : g1
                wire [7:0] data = data[15:8];
        end
  end
endgenerate

endmodule
```

## Actual Behavior

```-- Running command `read_verilog test2.v' --

1. Executing Verilog-2005 frontend: test2.v
Parsing Verilog input from `test2.v' to AST representation.
Generating RTLIL representation for module `\test2'.
test2.v:7: ERROR: Failed to detect width of signal access `\g0.data'!
```

## Expected Behavior

No error.

Like test1, iverilog has no issues with this, while verilator fails.

daveshah1, 2019-10-18:
Do you have a reference to one of the IEEE standards for the correct behaviour?

FWIW, Verific behaves the same way as Yosys - it produces this warning

    VERIFIC-WARNING [VERI-1216] oob.v:7: index 32 is out of range [31:0] for 'data'

and sets g1.data to 32'bx

tomverbeure, 2019-10-18:
> Do you have a reference to one of the IEEE standards for the correct behaviour?

I do not.

But the code example is a heavily reduced version of code that has been used for years, and successfully, with tools like ModelSim, VCS, DC, Quartus, Vivado, and others.

tomverbeure, 2019-10-18:
The reduced example was reduced to the point to where it reproduces the same error messages, but I didn't run it through one of the commercial tools. I may have reduced it too much.

So hold off until I've verified that.

tomverbeure, 2019-10-18:
I had changed the minimum example too much to protect the innocent. I have filed issue #1456 instead.

Closing this one.

closed 2019-10-18 by tomverbeure