YosysHQ/yosys #1458: VHDL to RTLIL or AST

moohtuh, 2019-10-19:
Hello!

Is there any way to get RTLIL or AST from VHDL sources? 

I looked at this: https://www.reddit.com/r/yosys/comments/c46qjc/vhdl_ast_representation_and_rtil_conversion/

Does anyone know if this is functional? 

Lastly, how do  I  join chat here : https://webchat.freenode.net/
what should be the channel ?

eddiehung, 2019-11-22:
CC @pepijndevos

pepijndevos, 2019-11-23:
https://github.com/tgingold/ghdlsynth-beta would be your best bet for reading VHDL into yosys.

On a related note, I'd really like to see this merged into yosys at some point.

The proper irc channels would be #yosys and ##openfpga , ghdl discussion happens on gitter.

umarcor, 2020-12-28:
FTR, see recent discussions in GHDL's repo about libghdl, the Python interface to libghdl, pyVHDLParser and [pyVHDLModel](https://github.com/VHDL/pyVHDLModel): https://github.com/ghdl/ghdl/labels/Interface%3A%20Python.

/cc @Paebbels

closed 2021-08-13 by mmicko