YosysHQ/yosys #1459: synth_ecp5 -abc9 hits assert in abc: Gia_ManInsertMfs: Assertion `iLitNew >= 0' failed.

antonblanchard, 2019-10-19:
## Steps to reproduce the issue

The following code:

```
module register_file(
    input wire clk,
    input wire write_enable,
    input wire [63:0] write_data,
    input wire [4:0] write_reg,
    input wire [4:0] read1_reg,
    output reg [63:0] read1_data,
    );

    reg [63:0] registers[0:31];

    always @(posedge clk) begin
      if (write_enable == 1'b1) begin
        registers[write_reg] <= write_data;
      end
    end

    always @(all) begin
      read1_data <= registers[read1_reg];
    end
endmodule
```

Hits an assert in abc:
```
# yosys -p "read_verilog register_file.v; synth_ecp5 -abc9"
```
```
ABC: yosys-abc: src/aig/gia/giaMfs.c:348: Gia_ManInsertMfs: Assertion `iLitNew >= 0' failed.
```

eddiehung, 2019-10-20:
This is a known issue with upstream ABC. Fix would be to do https://github.com/YosysHQ/yosys/commit/0acc51c3d82f65f73fa9e475c6fc41beabd925a6 for ecp5 too.

daveshah1, 2019-10-20:
Thanks, done in  fa989e5

closed 2019-10-20 by daveshah1

kost, 2020-04-21:
Not sure it is fixed:

```
4.48.9.4. Executing ABC9_EXE pass (technology mapping using ABC9).

4.48.9.5. Executing ABC9.
Running ABC command: <yosys-exe-dir>/yosys-abc -s -f <abc-temp-dir>/abc.script 2>&1
ABC: ABC command line: "source <abc-temp-dir>/abc.script".
ABC:
ABC: + read_lut <abc-temp-dir>/input.lut
ABC: + read_box <abc-temp-dir>/input.box
ABC: + &read <abc-temp-dir>/input.xaig
ABC: + &ps
ABC: <abc-temp-dir>/input : i/o =   6350/   7797  and =   38547  lev =   76 (6.46)  mem = 0.79 MB  box = 742  bb = 171
ABC: + &scorr
ABC: Warning: The network is combinational.
ABC: + &sweep
ABC: + &dc2
ABC: + &dch -f
ABC: + &ps
ABC: <abc-temp-dir>/input : i/o =   6350/   7797  and =   49999  lev =   34 (4.37)  mem = 0.92 MB  ch = 6483  box = 725  bb = 171
ABC: + &if -W 200 -v
ABC: K = 7. Memory (bytes): Truth =    0. Cut =   60. Obj =  140. Set =  636. CutMin = no
ABC: Node =   49999.  Ch =  5877.  Total mem =   10.84 MB. Peak cut mem =    0.43 MB.
ABC: P:  Del = 5004.00.  Ar =   40935.0.  Edge =    54903.  Cut =   606855.  T =     0.17 sec
ABC: P:  Del = 5004.00.  Ar =   40007.0.  Edge =    53607.  Cut =   593687.  T =     0.17 sec
ABC: P:  Del = 5004.00.  Ar =   18630.0.  Edge =    42446.  Cut =  1272596.  T =     0.32 sec
ABC: F:  Del = 5004.00.  Ar =   12868.0.  Edge =    37545.  Cut =  1186276.  T =     0.31 sec
ABC: A:  Del = 5004.00.  Ar =   12265.0.  Edge =    34799.  Cut =  1164408.  T =     0.57 sec
ABC: A:  Del = 5004.00.  Ar =   12173.0.  Edge =    34644.  Cut =  1166959.  T =     0.57 sec
ABC: Total time =     2.10 sec
ABC: + &mfs
ABC: Assertion failed: iLitNew >= 0 (src/aig/gia/giaMfs.c: Gia_ManInsertMfs: 388)
ABC: Aborted (core dumped)
ERROR: ABC: execution of command "/opt/ulx3s/bin/yosys-abc -s -f /tmp/yosys-abc-jHAbPF/abc.script 2>&1" failed: return code 134.
make: *** [makefile:41: bin/toplevel.json] Error 1
cp: cannot stat 'bin/toplevel.bit': No such file or directory
rm: cannot remove 'bin/toplevel.[bc]*': No such file or directory
ulx3s@eaf998bf9246:/scripts$ yosys -V
Yosys 0.9+2406 (git sha1 abc8f1fc, g++ 8.3.0 -Os)
```

daveshah1, 2020-04-21:
The pull request containing the new fix is #1974.

commit fa989e59 (2019-10-20): ecp5: Pass -nomfs to abc9

Fixes #1459
PR #1625 (2020-01-11): abc9: re-enable "&mfs" optimisation for synth_{xilinx,ecp5}
Since it was fixed upstream, and really fixes #1459