YosysHQ/yosys #1462: Xilinx_DSP pass throws std::out_of_range

andrewb1999, 2019-10-21:
## Steps to reproduce the issue
[bugreport.tar.gz](https://github.com/YosysHQ/yosys/files/3752776/bugreport.tar.gz)
Run the command in command.txt (after either setting YOSYS environment variable or editing the file).

## Expected behavior
Produce an output (without crashing) that contains DSPs.  We are not super concerned about the actual number of DSPs being produced at this point, and are not very sure about the exact number of DSPs that should be produced.

## Actual behavior
25.18. Executing XILINX_DSP pass (pack resources into DSPs)
...
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 0) >= this->size() (which is 0)
[1]    6930 abort (core dumped)  /scratch/safe/SymbiFlow/yosys.fresh/yosys -E .deps -p "tcl synth_xilinx.tcl"

mwkmwkmwk, 2019-11-18:
Testcase reduced to:

```
module top(...);
input wire [31:0] A;
output wire [31:0] P;

assign P = A * 32'h12300000;

endmodule
```

eddiehung, 2019-11-19:
Hi @andrewb1999. Thanks for the report, and sorry for the delay in getting back to you -- I've been on leave for the past few weeks. Please do keep reporting any issues you're coming across and I'll be much more responsive moving forward.

I can see that @mwkmwkmwk just got to this before I was back, and I can confirm that his #1499 PR gets past the problematic `xilinx_dsp` in your testcase until terminating with:
```
5.44. Executing CLKBUFMAP pass (inserting global clock buffers).
Inserting BUFG on leaf.clk_bft[0].
Inserting BUFG on leaf.leaf_interface_inst.sfc.ConCtrl.fifo1_inst.async_cmp.dirset_n[0].
Inserting BUFG on leaf.leaf_interface_inst.sfc.opc.output_port_cluster[0].OPort.fifo1_inst.async_cmp.dirset_n[0].
Inserting BUFG on leaf.leaf_interface_inst.sfc.opc.output_port_cluster[1].OPort.fifo1_inst.async_cmp.dirset_n[0].
ERROR: TCL interpreter returned an error: can't read "::env(symb_dir)": no such variable
```
but this seems to be a separate issue.

commit 15232a48 (2019-11-19): Fix #1462, #1480.
closed 2019-11-19 by mwkmwkmwk