YosysHQ/yosys #1464: tcl interpreter fails when including multiple dirs from enviro var

matthuszagh, 2019-10-24:
## Steps to reproduce the issue
$ cd /tmp && unzip bug.zip && cd /tmp/bug/sub/subdir && make
[bug.zip](https://github.com/YosysHQ/yosys/files/3765578/bug.zip)

## Expected behavior
Yosys should compile the 3 verilog files.

## Actual behavior
Yosys fails to find the include files. When only one include path is needed/specified this way, it works as expected. Also, as shown in the tcl script, if the -I paths are included directly (i.e. not via an environment variable) the script works. I've also included a puts in the tcl file, which, as far as I can tell, emits the right value.

I know ~nothing about tcl so if this is something that tcl is not meant to support please let me know. If that's the case, is there another way of passing include directories via makefile variables/environment variables that still permits running yosys via scripts (i.e. not yosys -p).

JakeMercer, 2019-11-02:
I _think_ it's due to it being passed as a single argument as opposed to being passed as separate arguments e.g. `-I/tmp/bug` and `-I/tmp/bug/sub/sub_other` vs. `-I/tmp/bug -I/tmp/bug/sub/sub_other` in which case it looks like only the _last_ directory is getting used.  I'm not sure if this is intentional or not but you can do the same command but placing `eval` in front which expands the variables in-place before evaluating the procedure:

```
eval read_verilog -sv $::env(VERILOG_INCS) $::env(TOP_MODULE_SRC)
```

matthuszagh, 2019-11-04:
@JakeMercer nice call, that works!

closed 2019-11-15 by whitequark