YosysHQ/yosys #1468: Cannot compile intentionally empty modules anymore in Yosys 0.9

norandomtechie, 2019-10-30:
## Steps to reproduce the issue

I have the following code in a file called test.v:

```
module top (hz100, reset, pb, ss7, ss6, ss5, ss4, ss3, ss2, ss1, ss0, left, right, red, green, blue);
  input hz100, reset;
  input [20:0] pb;
  output [7:0] ss7, ss6, ss5, ss4, ss3, ss2, ss1, ss0, left, right;
  output red, green, blue;
  
endmodule 
```

which is intended to be empty, since I use Yosys in another software which verifies if the code is syntactically correct. I attempted to compile this code with the command:

`yosys -p "synth_ice40 -top top -blif test.blif" test.v`

And the version of Yosys I'm using is Yosys 0.9+932 (git sha1 3c6a566d, gcc 6.3.0-18+deb9u1 -fPIC -Os).

## Actual behavior

yosys ends at the following step:

```
...
2.47. Printing statistics.

2.48. Executing CHECK pass (checking for obvious problems).
found and reported 0 problems.

2.49. Executing BLIF backend.
ERROR: Can't find top module `\top'!
```

## Expected behavior

This error did not happen in Yosys 0.8. It correctly compiled and produced a test.blif file as expected.

daveshah1, 2019-10-30:
Yosys has always turned empty modules into blackboxes automatically. The only change is that these are no longer written to the BLIF, as this was causing other issues.

To avoid this, there is a `-noblackbox` flag for the frontend. Use something like `read_verilog -noblackbox test.v; synth_ice40 -top top -blif test.blif`

norandomtechie, 2019-10-30:
> Yosys has always turned empty modules into blackboxes automatically. The only change is that these are no longer written to the BLIF, as this was causing other issues.
>
> To avoid this, there is a [...]

This worked for me. Thanks!

closed 2019-10-30 by norandomtechie