YosysHQ/yosys #1474: ERROR: Assert `lut_slacks[node] >= 0' failed in passes/techmap/flowmap.cc:1295.

ZirconiumX, 2019-11-05:
## Steps to reproduce the issue

Using 
[flowmap_1295.tar.gz](https://github.com/YosysHQ/yosys/files/3806602/flowmap_1295.tar.gz), run
`yosys -p "flowmap -maxlut 6 -optarea 4"  flowmap_1295.il`.

## Expected behavior

`flowmap` completes successfully.

## Actual behavior

```
2. Executing FLOWMAP pass (pack LUTs with FlowMap-r).
Labeling cells.
Mapped to 1 LUTs with maximum depth 1.

Optimizing area.
Relaxing with depth bound 1.
  Relaxed to 1 (+0) LUTs.
  Slack exhausted.
Relaxing with depth bound 2.
  Breaking LUT $auto$bugpoint.cc:222:simplify_something$247 to extract LUT $procmux$266.B_AND_S [10] (potential 150).
  Relaxed to 2 (+1) LUTs.
  Slack exhausted.
Relaxing with depth bound 3.
  Breaking LUT $auto$bugpoint.cc:222:simplify_something$247 to extract LUT \o_o [10] (potential 700).
  Breaking LUT $auto$bugpoint.cc:222:simplify_something$247 to extract LUT $auto$simplemap.cc:127:simplemap_reduce$342 [5] (potential 500).
  Breaking LUT $auto$bugpoint.cc:222:simplify_something$247 to extract LUT $auto$simplemap.cc:127:simplemap_reduce$351 [2] (potential 300).
  Breaking LUT $auto$bugpoint.cc:222:simplify_something$247 to extract LUT $auto$simplemap.cc:127:simplemap_reduce$356 [1] (potential 300).
  Breaking LUT $auto$simplemap.cc:127:simplemap_reduce$356 [1] to extract LUT $procmux$266.B_AND_S [10] (potential 150).
  Relaxed to 3 (+1) LUTs.
  Slack exhausted.
Relaxing with depth bound 4.
  Breaking LUT $auto$simplemap.cc:127:simplemap_reduce$356 [1] to extract LUT \o_o [10] (potential 700).
  Breaking LUT $auto$simplemap.cc:127:simplemap_reduce$356 [1] to extract LUT $auto$simplemap.cc:127:simplemap_reduce$342 [5] (potential 500).
  Breaking LUT $auto$simplemap.cc:127:simplemap_reduce$356 [1] to extract LUT $auto$simplemap.cc:127:simplemap_reduce$351 [2] (potential 300).
  Breaking LUT $auto$simplemap.cc:127:simplemap_reduce$351 [2] to extract LUT $procmux$266.B_AND_S [10] (potential 150).
  Relaxed to 4 (+1) LUTs.
  Slack exhausted.
Relaxing with depth bound 5.
  Breaking LUT $auto$simplemap.cc:127:simplemap_reduce$351 [2] to extract LUT \o_o [10] (potential 700).
  Breaking LUT $auto$simplemap.cc:127:simplemap_reduce$351 [2] to extract LUT $auto$simplemap.cc:127:simplemap_reduce$342 [5] (potential 500).
  Breaking LUT $auto$bugpoint.cc:222:simplify_something$247 to extract LUT $auto$simplemap.cc:256:simplemap_eqne$340 (potential 100).
  Breaking LUT $auto$simplemap.cc:256:simplemap_eqne$340 to extract LUT $procmux$266.B_AND_S [10] (potential 250).
  Breaking LUT $auto$simplemap.cc:256:simplemap_eqne$340 to extract LUT \o_o [10] (potential 900).
  Breaking LUT $auto$simplemap.cc:256:simplemap_eqne$340 to extract LUT $auto$simplemap.cc:127:simplemap_reduce$342 [5] (potential 700).
  Breaking LUT $auto$simplemap.cc:256:simplemap_eqne$340 to extract LUT $procmux$266.B_AND_S [26] (potential 100).
  Breaking LUT $auto$simplemap.cc:256:simplemap_eqne$340 to extract LUT \o_o [11] (potential 900).
  Breaking LUT $auto$simplemap.cc:127:simplemap_reduce$342 [5] to reuse LUT \o_o [11] (potential 450).
  Breaking LUT $auto$bugpoint.cc:222:simplify_something$247 to reuse LUT \o_o [11] (potential 500).
  Breaking LUT $auto$simplemap.cc:256:simplemap_eqne$340 to extract LUT $auto$simplemap.cc:127:simplemap_reduce$356 [1] (potential -100).
ERROR: Assert `lut_slacks[node] >= 0' failed in passes/techmap/flowmap.cc:1295.
```

cc @whitequark

whitequark, 2019-11-05:
Hm, why is it breaking a LUT with a negative potential? That implies the result would be worse.

whitequark, 2019-11-05:
Try this patch:
```diff
diff --git a/passes/techmap/flowmap.cc b/passes/techmap/flowmap.cc
index 5807178d..dae05a66 100644
--- a/passes/techmap/flowmap.cc
+++ b/passes/techmap/flowmap.cc
@@ -1188,7 +1188,8 @@ struct FlowmapWorker
 				if (debug_relax)
 					log("    Potential for breaking node %s: %d (Rex=%d, Rim=%d, Rslk=%d).\n",
 					    log_signal(lut_gate), p, r_ex, r_im, r_slk);
-				potentials[lut][lut_gate] = p;
+				if (p > 0)
+					potentials[lut][lut_gate] = p;
 			}
 		}
 	}
```
I'm not sure if it's the right fix though, would need to investigate in depth for that.

ZirconiumX, 2019-11-05:
Indeed, that fixes this (log below), along with #1405. However, it introduces a *new* bug (`Assert `node_origins.count(gate_node)' failed in passes/techmap/flowmap.cc:1369`) in #1475.

```
2. Executing FLOWMAP pass (pack LUTs with FlowMap-r).
Labeling cells.
Mapped to 1 LUTs with maximum depth 1.

Optimizing area.
Relaxing with depth bound 1.
  Relaxed to 1 (+0) LUTs.
  Slack exhausted.
Relaxing with depth bound 2.
  Breaking LUT $auto$bugpoint.cc:222:simplify_something$247 to extract LUT $procmux$266.B_AND_S [10] (potential 150).
  Relaxed to 2 (+1) LUTs.
  Slack exhausted.
Relaxing with depth bound 3.
  Breaking LUT $auto$bugpoint.cc:222:simplify_something$247 to extract LUT \o_o [10] (potential 700).
  Breaking LUT $auto$bugpoint.cc:222:simplify_something$247 to extract LUT $auto$simplemap.cc:127:simplemap_reduce$342 [5] (potential 500).
  Breaking LUT $auto$bugpoint.cc:222:simplify_something$247 to extract LUT $auto$simplemap.cc:127:simplemap_reduce$351 [2] (potential 300).
  Breaking LUT $auto$bugpoint.cc:222:simplify_something$247 to extract LUT $auto$simplemap.cc:127:simplemap_reduce$356 [1] (potential 300).
  Breaking LUT $auto$simplemap.cc:127:simplemap_reduce$356 [1] to extract LUT $procmux$266.B_AND_S [10] (potential 150).
  Relaxed to 3 (+1) LUTs.
  Slack exhausted.
Relaxing with depth bound 4.
  Breaking LUT $auto$simplemap.cc:127:simplemap_reduce$356 [1] to extract LUT \o_o [10] (potential 700).
  Breaking LUT $auto$simplemap.cc:127:simplemap_reduce$356 [1] to extract LUT $auto$simplemap.cc:127:simplemap_reduce$342 [5] (potential 500).
  Breaking LUT $auto$simplemap.cc:127:simplemap_reduce$356 [1] to extract LUT $auto$simplemap.cc:127:simplemap_reduce$351 [2] (potential 300).
  Breaking LUT $auto$simplemap.cc:127:simplemap_reduce$351 [2] to extract LUT $procmux$266.B_AND_S [10] (potential 150).
  Relaxed to 4 (+1) LUTs.
  Slack exhausted.
Relaxing with depth bound 5.
  Breaking LUT $auto$simplemap.cc:127:simplemap_reduce$351 [2] to extract LUT \o_o [10] (potential 700).
  Breaking LUT $auto$simplemap.cc:127:simplemap_reduce$351 [2] to extract LUT $auto$simplemap.cc:127:simplemap_reduce$342 [5] (potential 500).
  Breaking LUT $auto$bugpoint.cc:222:simplify_something$247 to extract LUT $auto$simplemap.cc:256:simplemap_eqne$340 (potential 100).
  Breaking LUT $auto$simplemap.cc:256:simplemap_eqne$340 to extract LUT $procmux$266.B_AND_S [10] (potential 250).
  Breaking LUT $auto$simplemap.cc:256:simplemap_eqne$340 to extract LUT \o_o [10] (potential 900).
  Breaking LUT $auto$simplemap.cc:256:simplemap_eqne$340 to extract LUT $auto$simplemap.cc:127:simplemap_reduce$342 [5] (potential 700).
  Breaking LUT $auto$simplemap.cc:256:simplemap_eqne$340 to extract LUT $procmux$266.B_AND_S [26] (potential 100).
  Breaking LUT $auto$simplemap.cc:256:simplemap_eqne$340 to extract LUT \o_o [11] (potential 900).
  Breaking LUT $auto$simplemap.cc:127:simplemap_reduce$342 [5] to reuse LUT \o_o [11] (potential 450).
  Breaking LUT $auto$bugpoint.cc:222:simplify_something$247 to reuse LUT \o_o [11] (potential 500).
  Relaxed to 4 (+0) LUTs.
  Slack exhausted.

Packing cells.
Packing alsru.$auto$simplemap.cc:136:simplemap_reduce$731.\Y (\o_o [11]).
  Packing alsru.$auto$simplemap.cc:85:simplemap_bitop$765.\Y ($procmux$266.B_AND_S [27]).
  Packed into a 1-LUT alsru.$auto$flowmap.cc:1411:pack_cells$1310.
Packing alsru.$auto$simplemap.cc:136:simplemap_reduce$348.\Y ($auto$simplemap.cc:127:simplemap_reduce$342 [5]).
  Packing alsru.$auto$simplemap.cc:136:simplemap_reduce$733.\Y (\o_o [10]).
  Packing alsru.$auto$simplemap.cc:85:simplemap_bitop$780.\Y ($procmux$266.B_AND_S [10]).
  Packing alsru.$auto$simplemap.cc:85:simplemap_bitop$764.\Y ($procmux$266.B_AND_S [26]).
  Packing alsru.$auto$simplemap.cc:85:simplemap_bitop$814.\Y (\$10 [10]).
  Packed into a 4-LUT alsru.$auto$flowmap.cc:1411:pack_cells$1311.
Packing alsru.$auto$simplemap.cc:136:simplemap_reduce$360.\Y ($auto$simplemap.cc:256:simplemap_eqne$340).
  Packing alsru.$auto$simplemap.cc:136:simplemap_reduce$358.\Y ($auto$simplemap.cc:127:simplemap_reduce$356 [1]).
  Packing alsru.$auto$simplemap.cc:136:simplemap_reduce$354.\Y ($auto$simplemap.cc:127:simplemap_reduce$351 [2]).
  Packed into a 4-LUT alsru.$auto$flowmap.cc:1411:pack_cells$1312.
Packing alsru.$auto$simplemap.cc:206:simplemap_lognot$362.\Y ($auto$bugpoint.cc:222:simplify_something$247).
  Packing alsru.$auto$simplemap.cc:85:simplemap_bitop$764.\Y ($procmux$266.B_AND_S [26]).
  Packed into a 2-LUT alsru.$auto$flowmap.cc:1411:pack_cells$1313.

Packed 12 cells (1 of them duplicated) into 4 LUTs.
Solution takes 118.8% of original gate area.
```

whitequark, 2019-11-12:
Unlike the other two issues you filed which have trivial fixes, this one points to a fundamental limitation of the IR used in the flowmap pass, and fixing it properly is fairly involved. In short, breaking out LUT w from LUT v and then LUT k from LUT v' does not necessarily result in a graph with LUTs v'', k, and w. In fact, if LUT k happens to include LUT w, then the second operation will result in a graph with only LUTs v'' and k. This violates the invariants of this operation and greatly confuses relaxation code. (It has nothing to do with negative potentials.)

This could probably be fixed relatively easily by going through any newly broken out LUT and "replaying" such previous operations. But it seems like there can be a more efficient way, perhaps with an IR that is more directly based on MFFCs? This is an interesting research problem worth maybe a week of work, which I don't have right now.

whitequark, 2019-11-12:
Incidentally the flowmap pass currently does not implement DF-Map, MP-Pack or Flow-Pack, so right now relaxation will always make the design perform worse.

ZirconiumX, 2019-11-12:
I understood most of your explanation here except for what an MFFC is in this context.

whitequark, 2019-11-12:
Minimal fanout-free cone. See the [long version of paper](https://sci-hub.se/10.1109/92.285741).

ZirconiumX, 2020-05-25:
So, I was examining this a bit further.

Here's the solution before relaxation:
[image]

The most important thing to note, in my opinion, is that this solution is area-optimal already, and `flowmap` should have returned as such: there is no way to relax the design without increasing the critical path.

At depth 2, `flowmap` decides to break out a gate:
```
    Considering breaking node $procmux$266.B_AND_S [10].
      Breaking requires a new LUT.
      Breaking eliminates LUT inputs $auto$bugpoint.cc:222:simplify_something$1210.
      Breaking may allow merging $auto$bugpoint.cc:222:simplify_something$1183 and $auto$bugpoint.cc:222:simplify_something$247.
      Breaking may allow merging $auto$bugpoint.cc:222:simplify_something$1190 and $auto$bugpoint.cc:222:simplify_something$247.
      Breaking may allow merging $auto$bugpoint.cc:222:simplify_something$1280 and $auto$bugpoint.cc:222:simplify_something$247.
      Breaking may allow merging $auto$bugpoint.cc:222:simplify_something$1197 and $auto$bugpoint.cc:222:simplify_something$247.
      Breaking may allow merging $auto$bugpoint.cc:222:simplify_something$1199 and $auto$bugpoint.cc:222:simplify_something$247.
      Breaking increases depth of LUT by 1.
      Breaking decreases slack of outputs $auto$bugpoint.cc:222:simplify_something$247.
    Potential for breaking node $procmux$266.B_AND_S [10]: 150 (Rex=-1, Rim=5, Rslk=1).
[snip]
Breaking LUT $auto$bugpoint.cc:222:simplify_something$247 to extract LUT $procmux$266.B_AND_S [10] (potential 150).
    Removing breaking gate $procmux$266.B_AND_S [10] from LUT.
    Removing LUT edge $auto$bugpoint.cc:222:simplify_something$1210 -> $auto$bugpoint.cc:222:simplify_something$247.
    Adding LUT edge $procmux$266.B_AND_S [10] -> $auto$bugpoint.cc:222:simplify_something$247.
  Updating slack and potentials.
    LUT $auto$bugpoint.cc:222:simplify_something$247 now has depth 2 and slack 0.
    LUT $procmux$266.B_AND_S [10] now has depth 1 and slack 0.
```
[image]

Here's the thing: this gate does *not* have slack, and `flowmap` knows it, but decides to break it out anyway. This suggests this code is incorrect:
https://github.com/YosysHQ/yosys/blob/89ed34fe55fd6a70b51ff68439e68f9a9e0b9e0f/passes/techmap/flowmap.cc#L1163-L1169

So to me, there are two interplaying bugs here: first that the relaxation code gets confused, as you've described, and second that `flowmap` should not attempt to relax this in the first place.

Ravenslofty, 2026-11-10:
This issue is moot now that flowmap got removed in #6061, so I'm going to close it.

closed 2026-11-10 by Ravenslofty