YosysHQ/yosys #1477: Plugins on Win32 platforms

kvide, 2019-11-06:
# Plugins on Win32 platforms

I would like to add support for plugins, for Yosys running on Win32 platforms.

Actually I already have a working version on MSYS2-64 with gdhlsynth as plugin, but it needs a bit of cleanup and some guidance from project owners on the preferred way to implement it.

## Linking plugins against yosys.exe

All non-member functions, structs and objects marked external in the header files in shared/include would need to be marked for export with `__declspec(dllexport)` or a platform dependent macro for that. Alternatively all functions in the exe could be exported with `--export_all_functions`.

Yosys.exe would need to be linked with:
`-Wl,--enable-auto-import -Wl,--out-implib,yosys_exe.a -Wl,--exclude-all-symbols`

The generated import library file should then be provided (via yosys-config) when linking plugins.

## Loading plugins on Win32

As debugging why a plugin isn't properly loaded is quite difficult without proper error messages, the [load_plugin()](https://github.com/YosysHQ/yosys/blob/master/passes/cmds/plugin.cc#L41) function in plugin.cc would need use native functions like ::LoadLibrary(), ::GetProcAddress() and::GetLastError();

From a win32 point of view, the correct way would be to provide a C function in the dll, that the exe could use to instantiate the struct derived from Pass parent (located in the exe).

```
#ifdef __MINGW32__
extern "C" __declspec(dllexport) Pass* __cdecl create_pass_instance()
{
	return new GhdlPass;
}
#endif
```

Then [Pass::run_register()](https://github.com/YosysHQ/yosys/blob/master/kernel/register.cc#L109) would use the pointer received from the dll for those Pass structures that are instantiated using heap and 'this' for those instantiated on the stack.

This concept is described in e.g. this article:

https://eli.thegreenplace.net/2011/09/16/exporting-c-classes-from-a-dll

#### Sidenote

The declarations of the structs derived from Pass have the same name as the struct type. 
```
struct VHDLDefaults : public Pass {...} VHDLDefaults;
```
IMHO that is not very elegant and I don't know of a way how to code wise specifically reference the struct type, as gcc seems to assume the declaration with the same name. It would be better if e.g. struct and class names are capitalized and their instances having the same name would begin with lowercase.

daveshah1, 2019-11-06:
As for (1), I would probably be in favour of exporting all symbols but you could add a YS_ macro to yosys.h similar to things like `YS_OVERRIDE` or `YS_NORETURN`.

For (2) worth noting there are already parts of Yosys that use native win32 functions for win32 builds like https://github.com/YosysHQ/yosys/blob/master/kernel/yosys.cc#L784-L806 so that would be fine. I don't have a particular problem with adding the function as described for Windows compatibility.

It's been a good few years since I've been doing Windows dev; but Yosys does support building with MSVC as well as mingw. With these changes; do you know if it be possible to add support for plugins for this option too?

kvide, 2019-11-06:
With proper compiler and link options it would work on MSVC as well. But yosys.exe and the plugins would need to be compiled using the same compiler, as the name mangling scheme is different between MSVC and GCC.

https://stackoverflow.com/questions/41524956/gcc-c-name-mangling-reference
https://stackoverflow.com/questions/48007224/msvc-function-name-mangling

mmicko, 2022-11-01:
Mingw support have been addressed by https://github.com/YosysHQ/yosys/pull/3534

closed 2022-11-01 by mmicko