YosysHQ/yosys #1478: How to generate post syntheses Area and timing report

n3donedo, 2019-11-07:
i have done all the steps as given in the documentation for the counter example. i want to know how to generate area and timing reports of this after syntheses. i am very new to this please specify each command and step to do so.
(this example may be very small for this type of question but may be useful in other circuits) 

##steps done
read_verilog counter.v
hierarchy -check -top counter
# the high-level stuff
proc; opt; fsm; opt; memory; opt
# mapping to internal cell library
techmap; opt
# mapping flip-flops to mycells.lib
dfflibmap -liberty mycells.lib
# mapping logic to mycells.lib
abc -liberty mycells.lib
# cleanup
clean

##outputs :
25.1.2. Re-integrating ABC results.
ABC RESULTS:              NAND cells:        3
ABC RESULTS:               NOR cells:        5
ABC RESULTS:               NOT cells:        2
ABC RESULTS:        internal signals:        6
ABC RESULTS:           input signals:        4
ABC RESULTS:          output signals:        2
Removing temp directory.

yosys> clean
Removed 0 unused cells and 8 unused wires.

##
need to find the area and timing report.

daveshah1, 2019-11-07:
`stat -liberty mycells.lib` will give you an area report. I'm less sure about timing but I think abc will generate one if you give it a delay target with -D.

n3donedo, 2019-11-09:
> `stat -liberty mycells.lib` will give you an area report. I'm less sure about timing but I think abc will generate one if you give it a delay target with -D.

closed 2019-11-09 by n3donedo