YosysHQ/yosys #1479: ERROR: Assert `current_val[i].wire != NULL || current_val[i] == value.bits[i]\' failed in ./kernel/consteval.h:79.

norandomtechie, 2019-11-07:
## Steps to reproduce the issue

The following code causes Yosys to fail an assertion error:

```
// Lab 11 
module top (hz100, reset, pb, ss7, ss6, ss5, ss4, ss3, ss2, ss1, ss0, left, right, red, green, blue);
  input hz100, reset;
  input [20:0] pb;
  output [7:0] ss7, ss6, ss5, ss4, ss3, ss2, ss1, ss0, left, right;
  output red, green, blue;
  // Variable, Wire, and Registries
  reg [31:0] entry; // value being entered
  reg [31:0] saved; // saved value to use as a second operand
  reg [4:0] op; // the selected operation
  reg displaysaved; // show entry or saved?
  reg equalpressed; // has equal been pressed?
  wire [4:0] key; // outkey of scankey module
  wire pressed; // strobe of scankey module
  wire [7:0] interss7;
  reg [31:0] nextresult;
  wire [31:0] addresult;
  wire [31:0] subresult;
  
  // Instantiations
  //fa u(.x(pb[0]), .y(pb[1]), .cin(pb[2]), .cout(right[1]), .s(right[0]));
  //fa4 u(.x(pb[7:4]), .y(pb[3:0]), .cin(pb[8]), .cout(right[4]), .s(right[3:0]));
  //fa8 u(.x(pb[15:8]), .y(pb[7:0]), .cin(pb[16]), .cout(red), .s(right[7:0]));
  //bcda8 u(.x(pb[15:8]), .y(pb[7:0]), .cin(pb[16]), .cout(red), .s(right[7:0]));
  scankey sk(.clk(hz100), .reset(reset), .inkeys({pb[19:15],5'b0,pb[9:0]}), .strobe(pressed), .outkey(key));
  bcdadd32 bcd8a(.x(saved), .y(entry), .s(addresult));
  bcdsub32 bcd8s(.x(saved), .y(entry), .s(subresult));
  
  // 7-Segment Set
  //wire [31:0] num = displaysaved ? saved : entry;
  wire [31:0] pos = displaysaved ? saved : entry;
  wire isneg = pos[31:28] == 9;
  wire [31:0] neg;
  bcdsub32 inverter(.x(0), .y(pos), .s(neg));
  wire [31:0] num = isneg ? neg : pos;
  assign ss7 = (isneg) ? 8'b01000000 : interss7;
  ssdec s7(.in(num[31:28]), .enable(|num[31:28]), .out(interss7));
  ssdec s6(.in(num[27:24]), .enable(|num[31:24]), .out(ss6));
  ssdec s5(.in(num[23:20]), .enable(|num[31:20]), .out(ss5));
  ssdec s4(.in(num[19:16]), .enable(|num[31:16]), .out(ss4));
  ssdec s3(.in(num[15:12]), .enable(|num[31:12]), .out(ss3));
  ssdec s2(.in(num[11:8]), .enable(|num[31:8]), .out(ss2));
  ssdec s1(.in(num[7:4]), .enable(|num[31:4]), .out(ss1));
  ssdec s0(.in(num[3:0]), .enable(1'b1), .out(ss0));
  
  // Main Body Code
  //always @(*) nextresult = entry;
  //always @(*) begin
  // nextresult <= saved;
  // if (op == 5'b00000) nextresult <= entry;
  // if (op == 5'b10010) nextresult <= saved + entry;
  // if (op == 5'b10011) nextresult <= saved - entry;
  //end
  always @(*) begin
    nextresult <= saved;
    if (op == 5'b00000) nextresult <= entry;
    if (op == 5'b10010) nextresult <= addresult;
    if (op == 5'b10011) nextresult <= subresult;
  end
  always @(posedge pressed, posedge reset) begin
    if (reset == 1'b1) begin
      entry <= 32'b0;
      saved <= 32'b0;
      op <= 5'b0;
      displaysaved <= 1'b0;
[... 310 lines cut ...]
  found state code: 5'10000
  found state code: 5'10001
  found state code: 5'10010
  found state code: 5'10011
  found ctrl output: $eq$test.v:56$13_Y
  found ctrl output: $eq$test.v:57$14_Y
  found ctrl output: $eq$test.v:58$15_Y
  ctrl inputs: { $eq$test.v:97$28_Y \key [4] \equalpressed \pb [19:15] \pb [9:1] }
  ctrl outputs: { $0\op[4:0] $eq$test.v:58$15_Y $eq$test.v:57$14_Y $eq$test.v:56$13_Y }
  transition:    5'00000 17'000-------------- ->    5'00000 8'00000001
  transition:    5'00000 17'001-------------- ->    5'00000 8'00000001
  transition:    5'00000 17'01--------------- ->    5'00000 8'00000001
  transition:    5'00000 17'10-00000000000000 ->    5'00000 8'00000001
  transition:    5'00000 17'10-00000000000001 ->    5'00001 8'00001001
  transition:    5'00000 17'10-0000000000001- ->    5'00010 8'00010001
  transition:    5'00000 17'10-000000000001-- ->    5'00011 8'00011001
  transition:    5'00000 17'10-00000000001--- ->    5'00100 8'00100001
  transition:    5'00000 17'10-0000000001---- ->    5'00101 8'00101001
  transition:    5'00000 17'10-000000001----- ->    5'00110 8'00110001
  transition:    5'00000 17'10-00000001------ ->    5'00111 8'00111001
  transition:    5'00000 17'10-0000001------- ->    5'01000 8'01000001
  transition:    5'00000 17'10-000001-------- ->    5'01001 8'01001001
  transition:    5'00000 17'10-00001--------- ->    5'01111 8'01111001
ERROR: Assert `current_val[i].wire != NULL || current_val[i] == value.bits[i]' failed in ./kernel/consteval.h:79.
```

ZirconiumX, 2019-11-08:
Minimised test case: using [1479.tar.gz](https://github.com/YosysHQ/yosys/files/3824757/1479.tar.gz), run `yosys -p "fsm" 1479.il`.

Unrelated to the bug, but I suggest you use `synth_ice40`'s `-abc9` option to improve performance.

mwkmwkmwk, 2019-11-08:
Another testcase, with a different pass:

```
module \top
  wire input 1 \A
  wire output 2 \Y
  cell $_AND_ \sub
    connect \A \A
    connect \B 1'0
    connect \Y \Y
  end
end
```

Blows up on `extract_fa`

mwkmwkmwk, 2019-11-08:
Nevermind, it's an unrelated bug...

norandomtechie, 2019-11-12:
I did try the abc9 option you specified, as "synth_ice40 -abc9 -top top", but it made no difference. It also had an error with another design, so I will have to hold off on using that option.

We think we may have fixed it by synchronizing outkey in the scankey module above to the given clock. Currently, outkey is assigned combinationally, but if we change the module to do the following:

```
module scankey (clk, reset, inkeys, strobe, outkey);
  input clk, reset;
  input [19:0] inkeys;
  output strobe;
  output reg [4:0] outkey;
  
  wire [4:0] next_outkey;

  reg [2:0] delay;
  always @(posedge clk, posedge reset)
  if (reset) begin
    delay <= 0;
    outkey <= 0;
  end
  else begin
    delay <= delay << 1 | |inkeys;
    outkey <= next_outkey;
  end

  assign strobe = delay[2];
  assign next_outkey = inkeys [19] ? 5'd19 :
                       inkeys [18] ? 5'd18 :
                       inkeys [17] ? 5'd17 :
                       inkeys [16] ? 5'd16 :
                       inkeys [15] ? 5'd15 :
                       inkeys [14] ? 5'd14 :
                       inkeys [13] ? 5'd13 :
                       inkeys [12] ? 5'd12 :
                       inkeys [11] ? 5'd11 :
                       inkeys [10] ? 5'd10 :
                       inkeys [ 9] ? 5'd9  :
                       inkeys [ 8] ? 5'd8  :
                       inkeys [ 7] ? 5'd7  :
                       inkeys [ 6] ? 5'd6  :
                       inkeys [ 5] ? 5'd5  :
                       inkeys [ 4] ? 5'd4  :
                       inkeys [ 3] ? 5'd3  :
                       inkeys [ 2] ? 5'd2  :
                       inkeys [ 1] ? 5'd1  : 5'd0;
endmodule
```

Then it works just fine. However, I still think this is a bug, because the design theoretically should still work, regardless of having it synchronized to the clock domain or not.

ZirconiumX, 2019-11-12:
This is undoubtedly a bug; compiler crashes are always bugs.

Your ABC9 error is weird; it generally produces better synthesis results
than the default, which is why I recommended it. However without further
information it's not really possible to diagnose it further.

Rodrigodd, 2024-04-03:
Stumble on the same assert. Was able to minimize my code by hand to the following:
```verilog
module test(input a, b, output c);
assign c = a ? 1 : 0;
assign c = b ? 1 : 0;
endmodule
```

Errors when I try to use `eval -table`:

```
$ yosys -p 'read_verilog test.v; eval -table a -table b'

 Yosys 0.38 (git sha1 543faed9c, gcc 13.2.1 -march=x86-64 -mtune=generic -O2 -fno-plt -fexceptions -fstack-clash-protection -fcf-protection -ffile-prefix-map=/build/yosys/src=/usr/src/debug/yosys -fPIC -Os)

-- Running command `read_verilog test.v; eval -table a -table b' --

1. Executing Verilog-2005 frontend: test.v
Parsing Verilog input from `test.v' to AST representation.
Generating RTLIL representation for module `\test'.
Successfully finished Verilog frontend.

2. Executing EVAL pass (evaluate the circuit given an input).
ERROR: Assert `current_val[i].wire != NULL || current_val[i] == value.bits[i]' failed in ./kernel/consteval.h:79.
```

But if I try:

```verilog
module test(input a, b, output c);
assign c = a ? 1 : 0;
assign c = b;
endmodule
```

It works:

```
$ yosys -p 'read_verilog test.v; eval -table a -table b'

 Yosys 0.38 (git sha1 543faed9c, gcc 13.2.1 -march=x86-64 -mtune=generic -O2 -fno-plt -fexceptions -fstack-clash-protection -fcf-protection -ffile-prefix-map=/build/yosys/src=/usr/src/debug/yosys -fPIC -Os)

-- Running command `read_verilog test.v; eval -table a -table b' --

1. Executing Verilog-2005 frontend: test.v
Parsing Verilog input from `test.v' to AST representation.
Generating RTLIL representation for module `\test'.
Successfully finished Verilog frontend.

2. Executing EVAL pass (evaluate the circuit given an input).

  \b  \a |  \c
 --- --- | ---
 1'0 1'0 | 1'0
 1'0 1'1 | 1'0
 1'1 1'0 | 1'1
 1'1 1'1 | 1'1

End of script. Logfile hash: ace15c8f0e, CPU: user 0.02s system 0.00s, MEM: 20.82 MB peak
Yosys 0.38 (git sha1 543faed9c, gcc 13.2.1 -march=x86-64 -mtune=generic -O2 -fno-plt -fexceptions -fstack-clash-protection -fcf-protection -ffile-prefix-map=/build/yosys/src=/usr/src/debug/yosys -fPIC -Os)
Time spent: 51% 1x eval (0 sec), 48% 2x read_verilog (0 sec)
```