YosysHQ/yosys #1480: Assert `nusers(P.extract_end(i)) <= 1' failed in ./passes/pmgen/xilinx_dsp_pm.h:588

antonblanchard, 2019-11-09:
I'm hitting an assert with a 64x64 multiplier when synthesizing for a Xilinx target:

```
module multiply(
  input         clock,
  input  [63:0] io_a,
  input  [63:0] io_b,
  output [63:0] io_sum
);
  reg [63:0] pipe_a;
  reg [63:0] pipe_b;
  reg [127:0] _T;
  reg [127:0] _T_1;
  assign io_sum = _T_1[63:0];
initial begin
end
  always @(posedge clock) begin
    pipe_a <= io_a;
    pipe_b <= io_b;
    _T <= pipe_a * pipe_b;
    _T_1 <= _T;
  end
endmodule
```

```
yosys -p "synth_xilinx" multiply.v
```

```
ERROR: Assert `nusers(P.extract_end(i)) <= 1' failed in ./passes/pmgen/xilinx_dsp_pm.h:588.
```

eddiehung, 2019-11-19:
Confirm that @mwkmwkmwk's PR #1499 fixes this original test-case for me!

mwkmwkmwk, 2019-11-19:
Fixed in #1499

closed 2019-11-19 by mwkmwkmwk