YosysHQ/yosys #1481: mul2dsp.v doesn't handle variable-width DSP blocks

ZirconiumX, 2019-11-09:
(as this is a feature request I hope you won't mind me eschewing the normal issue template).

`mul2dsp.v` is useful for converting a variable-width `$mul` cell into fixed-width DSP blocks. However, for my target FPGA family, the Cyclone V, the DSP blocks are either 9x9, 18x18 or 27x27. While it *is* feasible to map to an intermediate cell that gets lowered to a multiplier of sufficient width, it would likely be more efficient for `mul2dsp.v` to understand the range of DSP sizes available before picking one.

eddiehung, 2019-11-09:
Interesting use case I hadn't considered.

One seed idea to start building with would be to have three calls to `techmap ... mul2dsp.v` and use the `DSP_?_MINWIDTH` feature: https://github.com/YosysHQ/yosys/blob/master/techlibs/common/mul2dsp.v#L68-L75

This is currently used to ensure that we use soft logic for really small multipliers, but could forseeably be used for variable width too...?

ZirconiumX, 2019-11-09:
So you suggest trying to map each size of multiplier from largest to smallest using constrained sizes? That could work.

I think I need a benchmark for this, though; would marlann work?

eddiehung, 2019-11-09:
MARLANN probably wouldn't be very good because it is heavily tuned for the iCE40 and probably wouldn't be very efficient on the Cyclone V... I haven't dived deep into other benchmarks to know if we have any better ones on hand. Perhaps @dh73 has a better idea?