YosysHQ/yosys #1482: Optimisations in OPT_LUT pass

mo101, 2019-11-10:
Using the command line

`yosys -p synth_ice40 relut.v`

this code:

```
module top (input clk,output reg[3:0]z,input[3:0]x,input[3:0]y,input s);
  always @(posedge clk) begin
    z<=x+(s?y:0); // 7 LUTs
  end
endmodule
```

is synthesised differently from this code:

```
module top (input clk,output reg[3:0]z,input[3:0]x,input[3:0]y,input s);
  always @(posedge clk) begin
    z<=s?(x+y):x; // 4 LUTs
  end
endmodule
```

the second style being much more efficient as the OPT_LUT pass succeeds in packing all the logic into one LC per bit.

One result of this is that multipliers synthesised from LUTs, which as far as I can tell are essentially constructed using the first style, are much bigger than they need to be.

I am using Yosys 0.9+932 (git sha1 65f197e2, clang 6.0.0-1ubuntu2 -fPIC -Os).

whitequark, 2019-11-10:
I've added the `opt_lut` pass. Its original goal was to fold multiplexers into adders on iCE40, exactly as your second snippet shows. In your first snippet, the pass is inhibited by the presence of dedicated carry logic, since it is not whitebox-aware. It would be nontrivial to add whitebox awareness to `opt_lut` and IMO, LUT mapping in general should be whitebox-aware, therefore obsoleting simplistic passes like `opt_lut`.

mo101, 2019-11-10:
Thank you for the quick reply. I did glance at the code that implements the OPT_LUT pass and indeed it didn't look like a straightforward change.

It is a nevertheless a pity that LUT-based multipliers end up being about 30% bigger than they need to be. It looks like it should be possible to modify the TECHMAP pass so that the first layer of adders in the tree that sums the partial products takes advantage of what OPT_LUT can do, but I admit that is not a particularly clean solution.