YosysHQ/yosys #1491: Yosys unable to synthesize shift register

norandomtechie, 2019-11-14:
## Steps to reproduce the issue

Given a file test.v:

```
module top (hz100, reset, pb, ss7, ss6, ss5, ss4, ss3, ss2, ss1, ss0,
            left, right, red, green, blue);
  input hz100, reset;
  input [20:0] pb;
  output [7:0] ss7,ss6,ss5,ss4,ss3,ss2,ss1,ss0;
  output [7:0] left, right;
  output red, green, blue;
  
  crop c1 (pb[1], reset, ss7);
  
endmodule

module crop (collide, reset, crop);
  input wire collide;
  input wire reset;
  output reg [7:0] crop;
  
  always @(posedge collide, posedge reset)
    if (reset)
      crop <= 8'hff;
    else
      crop <= crop >> 1;
  
endmodule
```

The file was synthesized and a BLIF was produced using the following command:
yosys -p "synth_ice40 -abc9 -blif test.blif" test.v

The version of Yosys used was Yosys 0.9+932 (git sha1 3c6a566, gcc 6.3.0-18+deb9u1 -fPIC -Og).

## Expected behavior

Upon asserting `reset`, the value of `ss7` is supposed to be set to `8'hff`, and I can shift out bits to the right by asserting `pb[1]`.

## Actual behavior

`ss7` remains 0.

Upon further inspection of the BLIF file, it seems that the flip flops in my design are being ignored. From the statistics printed out by Yosys:

```
3.48. Printing statistics.

=== top ===
   Number of wires:                 19
   Number of wire bits:            116
   Number of public wires:          19
   Number of public wire bits:     116
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                  0
```

...when there should be at least 8 SB_DFFRs to implement the shift register in the crop module.

It works if I do `crop <= crop - 1;` and `crop <= crop << 1`, `just not `crop <= crop >> 1`.

whitequark, 2019-11-14:
It looks like `wreduce` removes all FFs:
```
2.11. Executing WREDUCE pass (reducing word size of cells).
Removed top 1 bits (of 8) from FF cell top.$techmap\c1.$procdff$11 ($adff).
Removed top 1 bits (of 7) from FF cell top.$techmap\c1.$procdff$11 ($adff).
Removed top 1 bits (of 6) from FF cell top.$techmap\c1.$procdff$11 ($adff).
Removed top 1 bits (of 5) from FF cell top.$techmap\c1.$procdff$11 ($adff).
Removed top 1 bits (of 4) from FF cell top.$techmap\c1.$procdff$11 ($adff).
Removed top 1 bits (of 3) from FF cell top.$techmap\c1.$procdff$11 ($adff).
Removed top 1 bits (of 2) from FF cell top.$techmap\c1.$procdff$11 ($adff).
Removed cell top.$techmap\c1.$procdff$11 ($adff).
```

whitequark, 2019-11-14:
Oh oops, looks like @daveshah1 managed to fix it while I got distracted triaging the issue.

PR #1492 (2019-11-17): wreduce: Don't trim zeros or sext when not matching ARST_VALUE
Fixes #1491
closed 2019-11-17 by cliffordwolf