YosysHQ/yosys #1493: RTLIL doesn't differentiate between simple bits and [0:0]

daveshah1, 2019-11-14:
Consider:
```verilog
module top(input [0:0] a, input b);
endmodule
```
which produces the following RTLIL
```
autoidx 1

attribute \blackbox 1
attribute \cells_not_processed 1
attribute \src "portstyles.v:1"
module \top

  attribute \src "portstyles.v:1"
  wire input 2 \b

  attribute \src "portstyles.v:1"
  wire input 1 \a
end
```
in which there is no difference between `a` and `b` (same applies, more relevantly in the below example, to downstream outputs like JSON too)

However, were I writing pin constraints, I would probably expect to constrain `a[0]` but `b` and see these names printed in output, etc. For the former case; aliases `a[0]` to `a` in pin constraints would be one solution on the nextpnr side; but it's worth considering if they could be differentiated - although it seems useless at first glance, it might matter to some generic patterns.

whitequark, 2019-11-15:
I would argue that aliasing it in nextpnr is *the* correct way to do it, because when writing generic code, I specifically want to be able to refer to the 1st bit of `a` as `a[0]` regardless of how wide `a` is.

cliffordwolf, 2019-11-17:
I agree with whitequark. `a`, `b`, `a[0]`, and `b[0]` should all be fine in this case. I think `input x` and `input [0:0] x` should be treated equivalent in yosys for all intents and purposes.

closed 2019-11-17 by daveshah1