YosysHQ/yosys #1512: Customizing the stat command

mickey4u, 2019-11-21:
I am new to yosys. However, I want to find out how to customize the stat command to print only the gate counts to a specified file. Or if there is already a way to do that, can be directed on how to achieve this.

daveshah1, 2019-11-22:
You can use `tee` to write the output from `stat` to a file. It doesn't support printing just gate counts.

mickey4u, 2019-11-22:
@daveshah1 I tried `stat tee -a info.txt`, and it printed something other than the stat information.

eddiehung, 2019-12-12:
`tee`'s help is:
```
yosys> help tee

    tee [-q] [-o logfile|-a logfile] cmd

Execute the specified command, optionally writing the commands output to the
specified logfile(s).

    -q
        Do not print output to the normal destination (console and/or log file).

    -o logfile
        Write output to this file, truncate if exists.

    -a logfile
        Write output to this file, append if exists.

    +INT, -INT
        Add/subtract INT from the -v setting for this command.
```

Thus you would want: `tee -a info.txt stat`

mickey4u, 2019-12-12:
@eddiehung Thanks

closed 2019-12-12 by mickey4u