YosysHQ/yosys #1532: Support Xilinx XC9500XL CPLD series

spth, 2019-11-26:
The Xilinx XC9500XL series is a popular CPLD series. It's 5V-tolerance, relatively low price, and many I/O pins make it a good choice for many projects.

IMO, this would be a good candiate for a new device to support by a free toolchain.

Xilinx is killing off its own non-free XC9500XL toolchain: The XC9500XL is only supported by older versions of the Xilinx WebISE tool, which itself is already deprecated in favour of the newer Vivado.

The XC9500XL being a CPLD series, not an FPGA series resulting in relative simplicity should make it easier to reverse-engineer and to target in synthesis.

whitequark, 2019-11-26:
Perhaps surprisingly, it can be harder to target CPLDs with Yosys because they use a sum-of-products architecture for logic cells, and most logic synthesis implemented so far is written with a LUT target in mind.

daveshah1, 2019-11-26:
To add to that, #519 contains some of the issues with real-world CPLD synthesis in the current state of Yosys. I suspect that many of them would apply to the XC9500XL also.