YosysHQ/yosys #1543: ice40_wrapcarry does not appear to respect (* keep *) attribute on SB_CARRY cells

weatherhead99, 2019-12-03:
## Steps to reproduce the issue

I am attempting to do experiments with tapped delay lines in ice40. I am using (for example)  a delay element as in the attached file delay_element.v. Note that I have applied (* keep *) attributes to all the cells which I need to remain there throughout synthesis.

In yosys I run the script

```
 read_verilog delay_line.v
synth_ice40
stat

   Number of wires:                  9
   Number of wire bits:              9
   Number of public wires:           9
   Number of public wire bits:       9
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                  1
     SB_DFFER                        1

```

As can be seen, the synthesis has optimised away the SB_CARRY (and hence also the SB_LUT4) that I explicitly need to be kept.

Having investigated in a little more detail, when the synth_ice40 command executes the ice40_opt pass, it is optimising away the $__ICE40_CARRY_WRAPPER cell (see int_4.v for netlist after ice40_opt) which was generated by ice40_wrapcarry (see the file int_3.v for netlist after ice40_wrapcarry). 

I think the fault lies in ice40_wrapcarry, as can be seen from int_3.v, there is no (* keep *) attribute on the $__ICE40_CARRY_WRAPPER  cell. I'm not familiar enough with yosys internals to know if this is actually the problem, though.

[delayline.zip](https://github.com/YosysHQ/yosys/files/3914011/delayline.zip)

## Expected behavior

synthesized netlist contains an SB_CARRY cell, which was marked as (* keep *)

## Actual behavior

synthesized netlist contains only SB_DFF** cell, SB_LUT4 and SB_CARRY have been optimised away.

eddiehung, 2019-12-03:
Thanks for bringing this oversight to my attention! #1545 should fix it, let me know how you get on.

weatherhead99, 2019-12-03:
hi @eddiehung,
thanks so much for getting to this so quickly! 
I haven't had a chance to run full simulations, but now synthesizing my simple delay line (see attached) actually looks like a delay line, and appears to have carry chain exactly as expected. So I do think this fixes it!
[image]

closed 2019-12-06 by weatherhead99

weatherhead99, 2019-12-08:
slight update on this after more experiments, it seems to fix all the problems I was having (other than other ones of my own making). The test also seems sensible though could I think be simplified to "just" an explicitly instantiated SB_CARRY if you want. 

Thanks so much for doing this and hope it makes it into a yosys release soon.

eddiehung, 2019-12-10:
> slight update on this after more experiments, it seems to fix all the problems I was having (other than other ones of my own making). The test also seems sensible though could I think be simplified [...]

No problem! #1545 has been merged into master now. Please double check it's still working for you and let me know otherwise. Thanks, and please do file issues for anything else you come across, or even to start contributing!

PR #1545 (2019-12-10): Preserve SB_CARRY name and attributes when using $__ICE40_CARRY_WRAPPER
Fixes #1543 

The only thing that isn't preserved is the SB_LUT4's name or attributes... I don't think this is critical right now though.