YosysHQ/yosys #1552: feedback loop missing in logic cell.

BigET, 2019-12-04:
## Steps to reproduce the issue

Yosys 0.9+932 (git sha1 fcce9401, clang 9.0.0 -fPIC -Os)

input : top.v

```
module top (
    output PIN_21, PIN_22, PIN_23, PIN_24, USBPU,
    input CLK, PIN_1, PIN_2, PIN_3, PIN_4, PIN_5, PIN_6, PIN_7, PIN_8, PIN_9, PIN_10, PIN_11, PIN_12, PIN_13
);

    reg compareBuf = 0;
    assign PIN_21 = compareBuf;

    always @(posedge CLK) begin
        compareBuf <= IN_1 < PIN_2 ? 1 : PIN_1 > PIN_2 ? 0 : compareBuf;
    end

endmodule

```

## Expected behavior

Have one LC with 3 inputs, the 2 from the pins and a third is a feedback loop.

## Actual behavior

There are 2 LCs:

1. has the LUT set to `PIN_1 < PIN_2` and has the dff enable.

2. has the LUT set to `PIN_1 ^ PIN_2` and is affecting the enable of the dff from the first LC.

So the expected and actual behavior have the same outcome, but is there a way to tell yosys to favor size vs speed?

It is arguable if the 2 LC solution is faster then the 1 LC, but for sure is bigger.

mwkmwkmwk, 2019-12-04:
Please specify your synthesis target, preferably by including the yosys script (or command line) you're using.

However, in general: we don't have a way to pick optimization target right now (and we're not particularly good about optimizing for either of them, anyhow).  As far as I understand, you don't have any actual bugs here, just an enhancement request?

BigET, 2019-12-04:
I do synthesis for TinyFPGA BX target.

I run:

```
yosys -p 'synth_ice40 -top top -blif top.blif' top.v
arachne-pnr -d 8k -P cm81 -o top.asc -p pins.pcf top.blif
```

If yosys chooses to use 2 LCs just because it has the space then you can call it an enhancement the posibiliti to give a "-Os" equivalent flag,

If yosys by doing this will end up exceeding the LC limit and fail to compile then I would call it a bug.

Giving that if I use the SB_* primitives yosys would do the thing I want, Let's say it is an enhancement.

eddiehung, 2020-01-14:
`compareBuf <= IN_1 < PIN_2 ? 1 : PIN_1 > PIN_2 ? 0 : compareBuf;`
is missing a `P`:
`compareBuf <= PIN_1 < PIN_2 ? 1 : PIN_1 > PIN_2 ? 0 : compareBuf;`

In this particular case, I believe what Yosys is doing is that it is inferring the SB_DFFE early in the flow, with the good intentions that doing so can absorb a mux from the design. Once this decision has been made, there's no going undoing it, which means that we have two inputs to drive (`D` and `E`) each of which must require at least one LUT.
A possible enhancement would be to have an `ice40_dffopt` similar to [`xilinx_dffopt`](#1557) to do some recovery, though I am still looking for more elegant solutions.