YosysHQ/yosys #1556: tests: efinix: mux16 case in mux.ys fails due to wrong number of LUT4 on macOS

kowalewskijan, 2019-12-06:
**Environment:**
- macOS Sierra (10.12.6)
- clang (8.1.0)
- iverilog (7f95abc)
- Yosys (7dece795)

Yosys built using official instructions from Readme.

**Test run on macOS:**
```
cd tests/arch/efinix
bash run_test.sh
```
**Error log:**
```
Running mux.ys..
ERROR: Assertion failed: selection contains 11 elements instead of the asserted 12: t:EFX_LUT4
Selection contains:
mux16/$abc$653$auto$blifparse.cc:498:parse_blif$654
mux16/$abc$653$auto$blifparse.cc:498:parse_blif$655
mux16/$abc$653$auto$blifparse.cc:498:parse_blif$656
mux16/$abc$653$auto$blifparse.cc:498:parse_blif$657
mux16/$abc$653$auto$blifparse.cc:498:parse_blif$658
mux16/$abc$653$auto$blifparse.cc:498:parse_blif$659
mux16/$abc$653$auto$blifparse.cc:498:parse_blif$660
mux16/$abc$653$auto$blifparse.cc:498:parse_blif$661
mux16/$abc$653$auto$blifparse.cc:498:parse_blif$662
mux16/$abc$653$auto$blifparse.cc:498:parse_blif$663
mux16/$abc$653$auto$blifparse.cc:498:parse_blif$664
make: *** [run-mux.ys] Error 1
```

I also ran the test on Linux and the error did not occur.
In attached blif files you can see that indeed there is a difference in number of LUTs.

[blif_files.zip](https://github.com/YosysHQ/yosys/files/3931483/blif_files.zip)

mmicko, 2019-12-06:
Thank you for reporting this issue. Was able to reproduce it on macOS machine.

eddiehung, 2019-12-20:
Just in case it gets lost, there was some discussion in https://github.com/YosysHQ/yosys/pull/1559#issuecomment-565581875 on the root cause:

> After much debugging with @mmicko it turns out the destruction order of STL containers in macOS is different to gcc/clang/VS: [apple/foundationdb#1862](https://github.com/apple/foundationdb/issues/1862)
> 
> Thus containers with IdString's in are destructed in a different order on macOS, and thus get recycled in a different order. And when other containers sort on the IdString index, we then get inconsistent behaviour.
> 
> ~Is there any good reason why we should be recycling 32b indices in the first place?~
> EDIT: I think the reason is that they are used to index into a `std::vector`...
> https://github.com/YosysHQ/yosys/blob/9ab1feeaf11adb6b675ac4034e246cb137d07db9/kernel/rtlil.h#L82
> 
> Alternatives include sorting the recycle list (at some cost, and even then recycling only happens at a `checkpoint()` which is only called at the end of every pass):
> 
> https://github.com/YosysHQ/yosys/blob/9ab1feeaf11adb6b675ac4034e246cb137d07db9/kernel/rtlil.h#L80

mmicko, 2020-01-29:
Fixed with  #1559

closed 2020-01-29 by mmicko