YosysHQ/yosys #1561: synth_ice40 -abc9 fails assertion `(int)Aig_Regular(pObjNew)->Level <= Required'

eddyb, 2019-12-08:
## Steps to reproduce the issue

See [`top.il` + `top.ys`](https://gist.github.com/eddyb/d957f88bee75a8fffde3fb9b4e1781a7).

> Also, make sure that the issue is actually reproducable in current git
master of Yosys.

**TODO**: check this (only tried it on `Yosys 0.9+932 (git sha1 3c41599ee1, g++ 8.3.0 -fPIC -Os)`)

## Expected behavior

Should succeed, worst case with an error because it can't fit the 24-bit division by 3, into an `iCEstick`.
Smaller sizes, such as 17-bit or 20-bit do succeed and perform correctly AFAICT, with `-abc9`.

## Actual behavior

Crashes with:
```
ABC: yosys-abc: src/opt/dar/darRefact.c:589: Dar_ManRefactor:
    Assertion `(int)Aig_Regular(pObjNew)->Level <= Required' failed`.
```

eddiehung, 2019-12-08:
Can repro on master (ecb0c68f). Am suspecting an upstream abc issue...

eddiehung, 2019-12-08:
@eddby Actually before I report this upstream, can you try and minimise the number of `SB_CARRY` cells (there are 915 in your example!) and/or share the original source for this `*.il`?
I understand from your edit that more bits doesn't equate to sustained breakage.. but can we reduce the width/depth? I had a go at minimising the `*.il` by deleting some output ports to no avail...

eddyb, 2019-12-09:
@eddiehung I'm not sure I can reduce this, as decreasing the main parameter (number of decimal digits, which leads to the bitwidths I mentioned before), or removing the division by 3, results in success.

[Here's the source `aoc-2019-1.py`](https://gist.github.com/eddyb/e3afeb490874f110b75f04cbd20dc20b/8ed86d3078af0051f3994beac2102773a1c1b6ed#file-aoc-2019-1-py), but note that it uses `nmigen` which outputs IL directly.
I had set the `NMIGEN_synth_opts=-abc9` environment variable to have yosys use `abc9`.

eddiehung, 2019-12-09:
Reported upstream: https://github.com/berkeley-abc/abc/issues/61

eddiehung, 2020-02-05:
Can still reproduce this on the latest master 5ebdc0f8

Ravenslofty, 2021-04-19:
It's now 2021 and I just hit this again while working on multiplier inference for QuickLogic...

Ravenslofty, 2021-04-19:
I attempted to reduce the testcase with `bugpoint` but to little avail. Still, having multiple testcases might help.

[dc2crash.tar.gz](https://github.com/YosysHQ/yosys/files/6336758/dc2crash.tar.gz)

Ravenslofty, 2021-04-20:
Some more information that might be useful: if I replace the timings in `qlal4s3_mult_cell_macro` with [this monstrosity I autogenerated a while back](https://github.com/antmicro/yosys/blob/ql_upstreaming_yosyshq/techlibs/quicklogic/pp3_cells_sim.v#L4862-L8797) it gets accepted, despite the netlists being identical.

eddiehung, 2021-08-31:
Can no longer reproduce with linked top.il + top.ys on b2e9717419e9a852f4e64f12891b8e9742900917. Closing for now.

closed 2021-08-31 by eddiehung