YosysHQ/yosys #1570: Fix mem_arst.v to be legal

wsnyder, 2019-12-12:
I was running some yosys tests on Verilator and the mem_arst.v test needs the patch attached to be applied to be legal.

You might also want to consider an enhancement to add a similar error message.

Thanks!

```
   Error: tests/simple/mem_arst.v:13: Duplicate declaration of signal: 'Data_o'
   note: ANSI ports must have type declared with the I/O (IEEE 2017 23.2.2.2)
     reg [DataWidth-1:0] Data_o;
  output [DataWidth-1:0] Data_o,
```

```
diff --git a/tests/simple/mem_arst.v b/tests/simple/mem_arst.v
index 9bd38fcb..88d0553b 100644
--- a/tests/simple/mem_arst.v
+++ b/tests/simple/mem_arst.v
@@ -7,11 +7,9 @@ module MyMem #(
   input                  Clk_i,
   input  [AddrWidth-1:0] Addr_i,
   input  [DataWidth-1:0] Data_i,
-  output [DataWidth-1:0] Data_o,
+  output reg [DataWidth-1:0] Data_o,
   input                  WR_i);

-  reg [DataWidth-1:0] Data_o;
-
   localparam Size = 2**AddrWidth;

   (* mem2reg *)

```

eddiehung, 2019-12-12:
Thanks for the patch @wsnyder! Am I right to understand that this is only necessary to be legal for the 2017 variant, and not prior variants?
Even so, there's no harm in applying it.

wsnyder, 2019-12-12:
All versions since 2001 which introduced port syntax.

eddiehung, 2019-12-12:
> All versions since 2001 which introduced port syntax.

Just for my understanding, I've seen much much code out there that does exactly this yet is accepted by all major front ends. Are they just being tolerant or am I missing something?

wsnyder, 2019-12-12:
This was/is still legal and you'll see in common code

    module x (y);
          output y;
          reg y;

This will give errors on several tools, and is IEEE illegal and is this issue:

    module x (
         output y);
         reg y;

PR #1571 (2019-12-19): mem_arst.v: do not redeclare ANSI port
Fixes #1570
closed 2019-12-19 by eddiehung