YosysHQ/yosys #1580: usage of windows-style absolute paths in -lut parameter of abc9 command

Chopper455, 2019-12-17:
## Steps to reproduce the issue

to use abc9-techmapping for ice40 or any other fpga with absolute path for lut file on windows:

> abc9 -W 400 -lut C:/.../ice40/abc_lp.lut -box C:/.../ice40/abc_lp.box

## Expected behavior

to map and not show any error

## Actual behavior

> 21. Executing ABC9 pass (technology mapping using ABC9).
> 
> 21.1. Extracting gate netlist of module `\NextZ80' to `<abc-temp-dir>/input.xaig'..
> ERROR: Abort in passes/techmap/abc9.cc:310.

I think it will be useful to be able to use absolute paths to specify lut and box files in abc9 command.
As it is now
```
			if (arg == "-lut" && argidx+1 < args.size()) {
				string arg = args[++argidx];
				size_t pos = arg.find_first_of(':');
				int lut_mode = 0, lut_mode2 = 0;
				if (pos != string::npos) {
					lut_mode = atoi(arg.substr(0, pos).c_str());
					lut_mode2 = atoi(arg.substr(pos+1).c_str());
				} else {
```
drive specification "C:/blablabla" is treated as specification of width interval.

eddiehung, 2019-12-18:
Thanks for the report @Chopper455! Can you try #1585 and see if that fixes it for you? Thanks!

Chopper455, 2019-12-20:
Sorry for the delay.
I've just tried out the pull request, it work works fine for me.
Thank you for your efforts:)

PR #1585 (2019-12-20): Interpret "abc9 -lut" as lut string only if [0-9:]
Fixes #1580
closed 2019-12-20 by eddiehung