YosysHQ/yosys #1597: yosys populating bogus LUT_INIT

BigET, 2019-12-27:
## Steps to reproduce the issue

Yosys 0.9+932 (git sha1 3e14ff16, gcc 6.3.0-18+deb9u1 -fPIC -Os)

Compile the next program, to reproduce put it in "top.v"

```
module top (
    input CLK, PIN_1, PIN_2, PIN_3, PIN_4, PIN_5,
        PIN_6, PIN_7, PIN_8, PIN_9, PIN_10, PIN_11, PIN_12, PIN_13, PIN_25,
    output USBPU, PIN_14, PIN_15, PIN_16, PIN_17, PIN_18,
        PIN_19, PIN_20, PIN_21, PIN_22, PIN_23, PIN_24,
);
    assign USBPU = 0;

    wire[5:0] parOut;
    wire[5:0] chrg;

    assign PIN_14 = parOut[0];
    assign PIN_15 = parOut[1];
    assign PIN_16 = parOut[2];
    assign PIN_17 = parOut[3];
    assign PIN_18 = parOut[4];
    assign PIN_19 = parOut[5];
    assign chrg[0] = PIN_3;
    assign chrg[1] = PIN_4;
    assign chrg[2] = PIN_5;
    assign chrg[3] = PIN_6;
    assign chrg[4] = PIN_7;
    assign chrg[5] = PIN_8;

    SSCounter6o sc6(PIN_1, CLK, PIN_2, PIN_9, chrg, parOut);

endmodule

module SSCounter6 (input wire rst, clk, adv, jmp, input wire [5:0] in, output reg[5:0] out);
    always @(posedge clk, posedge rst)
        if (rst) out <= 0;
        else if (adv || jmp) out <= jmp ? in : out + 1;
endmodule

// Optimized 6 bit counter, it should takes 7 cells.
/* b[5:1]                       /* b[0]
1010101010101010 in             1010101010101010 in
1100110011001100 jmp            1100110011001100 jmp
1111000011110000 loop           1111000011110000 loop
1111111100000000 carry          1111111100000000 -
----------------------          ----------------------
1000101110111000 out            1000101110001011 out
   8   B   B   8                   8   B   8   B
*/
module SSCounter6o (input wire rst, clk, adv, jmp, input wire [5:0] in, output wire[5:0] out);
    wire[4:0] co;
    wire[5:0] lo;
    wire ien;
    SB_LUT4 #(.LUT_INIT(16'hFFF0)) lien (ien, 0, 0, adv, jmp);
    SB_CARRY c0 (co[0], jmp, out[0], 1),
             c1 (co[1], jmp, out[1], co[0]),
             c2 (co[2], jmp, out[2], co[1]),
             c3 (co[3], jmp, out[3], co[2]),
             c4 (co[4], jmp, out[4], co[3]);
    SB_DFFER d0 (out[0], clk, ien, rst, lo[0]),
             d1 (out[1], clk, ien, rst, lo[1]),
             d2 (out[2], clk, ien, rst, lo[2]),
             d3 (out[3], clk, ien, rst, lo[3]),
             d4 (out[4], clk, ien, rst, lo[4]),
             d5 (out[5], clk, ien, rst, lo[5]);
    SB_LUT4 #(.LUT_INIT(16'h8B8B)) l0 (lo[0], in[0], jmp, out[0], 0);
    SB_LUT4 #(.LUT_INIT(16'h8BB8)) l1 (lo[1], in[1], jmp, out[1], co[0]);
    SB_LUT4 #(.LUT_INIT(16'h8BB8)) l2 (lo[2], in[2], jmp, out[2], co[1]);
    SB_LUT4 #(.LUT_INIT(16'h8BB8)) l3 (lo[3], in[3], jmp, out[3], co[2]);
    SB_LUT4 #(.LUT_INIT(16'h8BB8)) l4 (lo[4], in[4], jmp, out[4], co[3]);
    SB_LUT4 #(.LUT_INIT(16'h8BB8)) l5 (lo[5], in[5], jmp, out[5], co[4]);
endmodule
```

yosys -p 'synth_ice40 -top top -blif top.blif' top.v

## Expected behavior

The "SSCounter6o" should be an equivalent module to "SSCounter6" but just using 7 cells.

## Actual behavior

yosys mangles the bits in the .LUT_INIT for l0, l1, l2, l3, l4. Curiously l5 and lien are good. Look at the fragment from top.blif:

```
.gate SB_LUT4 I0=PIN_3 I1=PIN_9 I2=PIN_14 I3=$false O=sc6.lo[0]
.attr module_not_derived 00000000000000000000000000000001
.attr src "top.v:25|top.v:61|/usr/local/bin/../share/yosys/ice40/cells_map.v:56"
.param LUT_INIT 1111001100000011
.gate SB_LUT4 I0=PIN_4 I1=PIN_9 I2=PIN_15 I3=sc6.co[0] O=sc6.lo[1]
.attr module_not_derived 00000000000000000000000000000001
.attr src "top.v:25|top.v:62|/usr/local/bin/../share/yosys/ice40/cells_map.v:56"
.param LUT_INIT 1111011000000110
.gate SB_LUT4 I0=PIN_5 I1=PIN_9 I2=PIN_16 I3=sc6.co[1] O=sc6.lo[2]
.attr module_not_derived 00000000000000000000000000000001
.attr src "top.v:25|top.v:63|/usr/local/bin/../share/yosys/ice40/cells_map.v:56"
.param LUT_INIT 1111011000000110
.gate SB_LUT4 I0=PIN_6 I1=PIN_9 I2=PIN_17 I3=sc6.co[2] O=sc6.lo[3]
.attr module_not_derived 00000000000000000000000000000001
.attr src "top.v:25|top.v:64|/usr/local/bin/../share/yosys/ice40/cells_map.v:56"
.param LUT_INIT 1111011000000110
.gate SB_LUT4 I0=PIN_7 I1=PIN_9 I2=PIN_18 I3=sc6.co[3] O=sc6.lo[4]
.attr module_not_derived 00000000000000000000000000000001
.attr src "top.v:25|top.v:65|/usr/local/bin/../share/yosys/ice40/cells_map.v:56"
.param LUT_INIT 1111011000000110
.gate SB_LUT4 I0=PIN_8 I1=PIN_9 I2=PIN_19 I3=sc6.co[4] O=sc6.lo[5]
.attr module_not_derived 00000000000000000000000000000001
.attr src "top.v:25|top.v:66"
.param LUT_INIT 1000101110111000
.gate SB_LUT4 I0=$false I1=$false I2=PIN_2 I3=PIN_9 O=sc6.ien
.attr module_not_derived 00000000000000000000000000000001
.attr src "top.v:25|top.v:49"
.param LUT_INIT 1111111111110000
```

ZirconiumX, 2019-12-28:
`synth_ice40 -abc9 -top top`

```
=== top ===

   Number of wires:                 38
   Number of wire bits:             67
   Number of public wires:          38
   Number of public wire bits:      67
   Number of memories:               0
   Number of memory bits:            0
   Number of processes:              0
   Number of cells:                 18
     SB_CARRY                        5
     SB_DFFER                        6
     SB_LUT4                         7
```

This is what you're looking for, right?

ZirconiumX, 2019-12-28:
It's a bit unc;lear what you're asking. If you're complaining that the LUT masks don't match up, that's expected behaviour unless it's introduced a bug in your code.

BigET, 2019-12-28:
Please explain why the LUT mask for l1,l2,l3,l4 is 1111011000000110,

yet for l5 is 1000101110111000.

All these have the same LUT_INIT provided. 16'h8BB8 (which is 16'b1000101110111000).

eddiehung, 2020-01-13:
Hi @BIgET. So sorry for the inconvenience, and the time it took for me to get to it! It turns out that the `ice40_wrapcarry` pass that I created (a pass which wraps `SB_CARRY` and the associated `SB_LUT4` into an atomic entity) was not doing the right twiddling in order for some later untwiddling to happen.

The common flow is to use Yosys to automatically infer the LUT mask for carry chains, and in that flow it was behaving correctly because it is a twiddle-proof mask. However, the manual instantiation approach isn't guaranteed to be so and it was broken all this time. Thank you for bringing it to my attention, and please give #1628 a try. Your test-case now passes equivalence checking, whereas it didn't before.

eddiehung, 2020-01-13:
Also, it's worth bearing in mind that even manually instantiations of target primitives might still be modified by Yosys (under good pretences of trying to optimise in some dimension, even if it doesn't end up being so or hurts another dimension) unless a `(* keep *)` attribute is attached.
Not sure if this would have helped here since `ice40_wrapcarry` and/or `abc9` probably don't respect this attribute (yet)

PR #1660 (2020-01-28): Unpermute LUT ordering for ice40/ecp5/xilinx
Originally, `abc9` required that the inputs be ordered fastest first, and so I changed the `techmap`-ing rules for supported architectures to do this, including twiddling the LUT masks to cope.

I've just realised that there's no need to do that, as I can just reverse the ordering coming out of `abc9`... and this cleans up the `techmap` rules as well as makes debugging easier.

Fixes #1597 
Supersedes/fixes #1628
closed 2020-01-28 by eddiehung